autowarefoundation/AWSIM #7: Docker container for AWSIM

doganulus, 2022-10-14:
Is there any plans to package the simulator inside a docker container? 

It would be great if the simulator can be run as a container, especially together with prebuilt containers of Autoware.

mackierx111, 2022-10-15:
Thanks for the suggestion. Not planned yet, but a good idea.  I too would like to simplify the environment building process. Please give me some time to consider if I'm missing something.

tleyden, 2022-11-02:
I tried running running AWSim from within the autoware docker container, but was not able to get it to work.  See [this autoware discussion post](https://github.com/autowarefoundation/autoware/discussions/2991) for more details.

Is the recommended way to run autoware + AWSim to install autoware from source rather than running inside of a docker container?

mackierx111, 2022-11-08:
@tleyden 
Sorry for the delay in replying.
> Is the recommended way to run autoware + AWSim to install autoware from source rather than running inside of a docker container?

Yes, That is correct. We have not yet checked out running AWSIM with Docker.
If you follow this tutorial, you should be able to run AWSIM in a stable.
https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/

shmpwk, 2023-01-30:
@doganulus @tleyden 
Could you try this? https://github.com/tier4/AWSIM/issues/81

doganulus, 2023-01-30:
Why does #81 need to install AWSIM on an Autoware container?

shmpwk, 2023-01-30:
Yes it can but just he didn't.

mackierx111, 2023-06-19:
https://github.com/tier4/AWSIM/issues/81

doganulus, 2023-06-19:
Could you please open this issue? 

The issue #81 is not the solution for the feature here. This is about packaging the AWSIM as a container.

WfHit, 2023-07-29:
I just find a way to run AWSIM  in docker container: (Much thanks for the great job done by [rickyjames35](https://github.com/rickyjames35/vulkan_docker_test/tree/main) to solve the vulkan problem)

1. Download AWSIM and unzip the downloaded file

2. Create an image containing both vulkan and ros humble with the follow dockerfile
```dockerfile
FROM ros:humble-ros-base-jammy

# Needed to share GPU
ENV NVIDIA_DRIVER_CAPABILITIES=all
ENV NVIDIA_VISIBLE_DEVICES=all

RUN apt-get update && \
    export DEBIAN_FRONTEND=noninteractive && \
    apt-get -y install \
    pciutils \
    vulkan-tools \
    mesa-utils \
    ros-humble-rmw-cyclonedds-cpp
    
RUN echo "export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp" >> ~/.bashrc
```
3. Run a container with volume containing unzipped AWSIM folder
```
docker run --gpus all --runtime=nvidia -it --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix  -v <path to you AWSIM folder>:/AWSIM <your image name> /bin/bash
```
4.  In container, launch AWSIM
```
/AWSIM/AWSIM_demo.x86_64
```

yulong88888, 2023-08-04:
> I just find a way to run AWSIM in docker container: (Much thanks for the great job done by [rickyjames35](https://github.com/rickyjames35/vulkan_docker_test/tree/main) to solve the vulkan problem) [...]

I am very happy to see this solution, but I tried it on wsl2 of win11, and there is no error, but it also failed to start. May I ask what environment you are using it in?

WfHit, 2023-08-07:
> I am very happy to see this solution, but I tried it on wsl2 of win11, and there is no error, but it also failed to start. May I ask what environment you are using it in?

I run  it on ubuntu 2204 and not test on wsl2

oguzkaganozt, 2024-02-26:
> I just find a way to run AWSIM in docker container: (Much thanks for the great job done by [rickyjames35](https://github.com/rickyjames35/vulkan_docker_test/tree/main) to solve the vulkan problem) [...]

Recently tried with this Dockerfile and docker arguments and it is failing, also with autoware-universe:prebuilt image and it is the same. Has anyone tried again nowadays ?

doganulus, 2024-02-26:
I would like to have a standalone AWSIM container to work as a sidecar with Autoware runtime container and perhaps with others. My latest container was:

```dockerfile
FROM ubuntu:jammy as builder

ARG VERSION=v1.2.0
ARG AWSIM_VERSION=$VERSION
ARG AWSIM_RELEASE_PATH=https://github.com/tier4/AWSIM/releases/download/${AWSIM_VERSION}/AWSIM_${AWSIM_VERSION}.zip
ARG AWSIM_ADDITIONAL_MAPS=https://github.com/tier4/AWSIM/releases/download/${AWSIM_VERSION}/nishishinjuku_autoware_map.zip

RUN apt-get update && export DEBIAN_FRONTEND=noninteractive \
    && apt-get -y install \
        wget \
        unzip \
    && apt-get autoremove -y && apt-get clean -y && rm -rf /var/lib/apt/lists/* 

RUN wget -qO /tmp/awsim_release.zip ${AWSIM_RELEASE_PATH} && \
    unzip /tmp/awsim_release.zip -d /awsim && \
    rm /tmp/awsim_release.zip

FROM nvidia/cuda:12.2.0-runtime-ubuntu22.04 as runtime

ARG VERSION=v1.2.0
ARG AWSIM_VERSION=$VERSION

# RUN apt-get update && export DEBIAN_FRONTEND=noninteractive \
#     && apt-get -y install \
#         sudo \
#     && apt-get autoremove -y && apt-get clean -y && rm -rf /var/lib/apt/lists/* 

# RUN groupadd awsim -g 1000 \
#    && useradd -ms /bin/bash awsim -g 1000 -u 1000 \
#    && printf "awsim:awsim" | chpasswd \
#    && printf "awsim ALL= NOPASSWD: ALL\\n" >> /etc/sudoers

RUN apt-get update && export DEBIAN_FRONTEND=noninteractive \
    && apt-get -y install \
        x11-apps \
        x11vnc \
        xvfb \
    && apt-get autoremove -y && apt-get clean -y && rm -rf /var/lib/apt/lists/* 

COPY --from=builder --chown=root:root /awsim/AWSIM_${AWSIM_VERSION} /opt/awsim

# USER awsim
# WORKDIR /home/awsim/

ENV RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

CMD ["/opt/awsim/AWSIM_demo.x86_64"]

```

It builds but couldn't test it recently with a computer with Nvidia hardware. Last time I had got segmentation faults. I think some initialization work is needed for `x11` forwarding mechanisms in the eventual `entrypoint.sh`.

mitsudome-r, 2024-03-21:
I was able to run AWSIM using the following commands:

Download AWSIM and maps for AWSIM by following https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/

```
git clone git@github.com:autowarefoundation/autoware
cd autoware
./setup-dev-env.sh docker
./docker/run.sh --map-path /home/mitsudome-r/autoware_map/nishishinjuku_autoware_map/ --workspace /home/mitsudome-r/Downloads/AWSIM_v1.2.0 /bin/bash

# inside docker container
cd /workspace
sudo ./AWSIM_v1.2.0.x86_64
```

AWSIM failed with segmentation fault if I run it without sudo.
Also, I still get the following error messages (probably due to some access issues to GPU devices), but I was still able to launch it without segmentation faults.

```
MESA: error: Failed to query drm device.
glx: failed to create dri3 screen
failed to load driver: iris
failed to open /dev/dri/card0: No such file or directory
failed to load driver: iris
```

oguzkaganozt, 2024-03-28:
@mitsudome-r 
I tried the same 
```
git clone git@github.com:autowarefoundation/autoware
cd autoware
./setup-dev-env.sh docker
./docker/run.sh --map-path /home/mitsudome-r/autoware_map/nishishinjuku_autoware_map/ --workspace /home/mitsudome-r/Downloads/AWSIM_v1.2.0 /bin/bash

# inside docker container
cd /workspace
sudo ./AWSIM_v1.2.0.x86_64
```

 with **sudo** but I am getting segmentation fault. 

Can you try `nvidia-smi ` inside and outside of the container ? And also what is your exact GPU model and its driver in use ?

mitsudome-r, 2024-03-28:
@oguzkaganozt Here's the information

GPU: NVIDIA GeForce RTX 3060 Mobile 

`nvidia-smi` inside container:
```
Fri Mar 29 00:08:27 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 545.23.08              Driver Version: 545.23.08    CUDA Version: 12.3     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3060 ...    On  | 00000000:01:00.0  On |                  N/A |
| N/A   54C    P8              11W / 115W |     57MiB /  6144MiB |     18%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      2150      G   /usr/lib/xorg/Xorg                           45MiB |
+---------------------------------------------------------------------------------------+

```

outside container:
```
Fri Mar 29 00:07:28 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 545.23.08              Driver Version: 545.23.08    CUDA Version: 12.3     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3060 ...    On  | 00000000:01:00.0  On |                  N/A |
| N/A   60C    P3              17W / 115W |     57MiB /  6144MiB |      2%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
+---------------------------------------------------------------------------------------+
```

mitsudome-r, 2024-03-28:
I also tested with a machine with `GeForce GTX 1070` as well, and it worked.

mitsudome-r, 2024-03-28:
Could you try running the run.sh command with sudo and see if that changes anything as well?

oguzkaganozt, 2024-05-03:
It is now running with the new docker containers and AWSIM version 1.2.1. With these instructions:
```
git clone git@github.com:autowarefoundation/autoware
cd autoware
./setup-dev-env.sh docker
./docker/run.sh --map-path /home/mitsudome-r/autoware_map/nishishinjuku_autoware_map/ --workspace /home/mitsudome-r/Downloads/AWSIM_v1.2.0 /bin/bash

# inside docker container
cd /workspace
sudo ./AWSIM_v1.2.0.x86_64
```

doganulus, 2024-05-06:
Our solution is here:
https://github.com/bounverif/containers/tree/main/awsim-desktop

It is based on `ghcr.io/selkies-project/nvidia-egl-desktop`.