autowarefoundation/AWSIM #15: No Image from image_raw showing in Rviz

tomaslieven, 2022-10-25:
awsim will list all topics ie 

```
/clock
/control/command/control_cmd
/control/command/emergency_cmd
/control/command/gear_cmd
/control/command/hazard_lights_cmd
/control/command/turn_indicators_cmd
/parameter_events
/rosout
/sensing/camera/traffic_light/camera_info
/sensing/camera/traffic_light/image_raw
/sensing/gnss/pose
/sensing/gnss/pose_with_covariance
/sensing/imu/tamagawa/imu_raw
/sensing/lidar/top/pointcloud_raw
/sensing/lidar/top/pointcloud_raw_ex
/vehicle/status/control_mode
/vehicle/status/gear_status
/vehicle/status/hazard_lights_status
/vehicle/status/steering_status
/vehicle/status/turn_indicators_status
/vehicle/status/velocity_status

```

also 
/sensing/camera/traffic_light/image_raw

is echoing and streaming data:
```
---
header:
  stamp:
    sec: 386
    nanosec: 89991370
  frame_id: traffic_light_left_camera/camera_optical_link
height: 720
width: 1280
encoding: bgr8
is_bigendian: 0
step: 3840
data:
- 78
- 70
- 69
- 79
- 72
- 69
- 78
- 72
- '...'
---
```
but i cant get anything in Rviz or inside:
/perception/traffic_light_recognition/debug/rois

[image]

also topic /sensing/camera/traffic_light/camera_info seems ok
```
header:
  stamp:
    sec: 894
    nanosec: 289980011
  frame_id: traffic_light_left_camera/camera_optical_link
height: 720
width: 1280
distortion_model: plumb_bob
d:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
k:
- 365.71429443359375
- 0.0
- 640.5
- 0.0
- 365.4822082519531
- 360.5
- 0.0
- 0.0
- 1.0
r:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
p:
- 365.71429443359375
- 0.0
- 640.5
- 0.0
- 0.0
- 365.4822082519531
- 360.5
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
---
```

running ros galactic, tried main or even awsim-stable branches, also unity project all same results not processing camera stream

shmpwk, 2022-10-27:
1. Is there the topic `/perception/traffic_light_recognition/debug/rois` ?

2. Could you check the RViz setting of `/perception/traffic_light_recognition/debug/rois`
>                Depth: 5
>                Durability Policy: Volatile
>                History Policy: Keep Last
>                Reliability Policy: Best Effort

[image]

tomaslieven, 2022-10-27:
> @tomaslieven
> Thanks for the topic list. What are your PC specs?

running mobile RTX3060 with i7 just did clean install of Ubuntu 20.04 just to make sure im running all right frameworks

tomaslieven, 2022-10-27:
> @tomaslieven
> 
> 1. Is there the topic `/perception/traffic_light_recognition/debug/rois` ?
> 2. Could you check the RViz setting of `/perception/traffic_light_recognition/debug/rois` ?
> 
> > ```
> >            Depth: 5
> >            Durability Policy: Volatile
> >            History Policy: Keep Last
> >            Reliability Policy: Best Effort
> > ```
> 
> [image]

yes this are default settings within /perception/traffic_light_recognition/debug/rois , ros2 echo on topic is not showing anything

shmpwk, 2022-10-27:
@tomaslieven  Thank you for your checking. 
Could you share your log files of the console output under ~/.ros/log/[DATE, etc] (e.x. ~/.ros/log/2022-10-27-10-09-39-837180-npc2201036-2629664/).

shmpwk, 2022-10-27:
@tomaslieven 
Did you succeed the Autoware tutorial such as [planning simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/) and [rosbag replay simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/)?

tomaslieven, 2022-10-28:
sure attaching default launch log without any code changes, 

@shmpwk i completed planning, but because of perception not streaming image data had to set use_empty_dynamic_object_publisher to True and comment out some logic otherwise planning would not work, all other topics are streaming ok as image above showing with lidar or vehicle state data ie blinkers in sim

[launch.log](https://github.com/tier4/AWSIM/files/9885907/launch.log)

[image]

shmpwk, 2022-10-28:
@tomaslieven 
Could you try to add `pcl_ros` in `universe/autoware.universe/sensing/pointcloud_preprocessor/CMakeLists.txt` 
https://github.com/autowarefoundation/autoware.universe/pull/2101/files

tomaslieven, 2022-10-28:
thanks! this seems to resolve the issues, i stashed all the changes and added pcl_ros seems to work even without use_empty_dynamic_object_publisher

[image]

closed 2022-10-28 by tomaslieven

shmpwk, 2022-10-28:
Thank you for your checking. 
I will update the Autoware `awsim-stable` branch ASAP to reflect https://github.com/autowarefoundation/autoware.universe/pull/2101/files :bow:

shmpwk, 2022-10-28:
I updated the awsim-stable branch by https://github.com/autowarefoundation/autoware/pull/2982.