autowarefoundation/AWSIM #17: Issue about unsuccessful 2D pose estimate when work with autoware

hangger-258, 2022-10-27:
Although I install the AWSIM and Autoware follow the Quick start demo, In the STEP5. Let's Self-Driving simulation, Launch both the AWSIM and Autoware. 
I can't Automaticalely set 2D Pose Estimate with RViz as the tutorial.  The state panel show that it is always in INITIALIZING. And when I set the 2D goal post, there is no path to be generated.
Additionally, The autoware can work itself in the official tutorial of autoware.

shmpwk, 2022-10-27:
@hangger-258 
Could you share your log files of the console output under ~/.ros/log/[DATE, etc] (e.x. ~/.ros/log/2022-10-27-10-09-39-837180-npc2201036-2629664/)

shmpwk, 2022-10-27:
@hangger-258 
Thank you.
Looking at your [log file](https://drive.google.com/file/d/1Hm-534tlP94WgoIlQyq-GQqbq6QXHCVD/view?usp=sharing), there are several suspicious ERROR log.

For example, L24 says 
```
1666844014.6560595 [ERROR] [map_hash_generator-9]: process has died [pid 19716, exit code 1, cmd '/home/human/Desktop/AWSIM-autoware/autoware/install/map_loader/lib/map_loader/map_hash_generator --ros-args -r __node:=map_hash_generator -r __ns:=/map --params-file /tmp/launch_params_7z7p_cgj --params-file /tmp/launch_params_scsb65ac'].
```
It corresponds to L138-L155.
```
1666844014.7487721 [map_hash_generator-9] Traceback (most recent call last):
1666844014.7557201 [map_hash_generator-9]   File "/home/human/Desktop/AWSIM-autoware/autoware/install/map_loader/lib/map_loader/map_hash_generator", line 21, in <module>
1666844014.7580757 [map_hash_generator-9]     from rclpy.node import Node
1666844014.7582481 [map_hash_generator-9]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/node.py", line 43, in <module>
1666844014.7590582 [map_hash_generator-9]     from rclpy.client import Client
1666844014.7591376 [map_hash_generator-9]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/client.py", line 22, in <module>
1666844014.7593884 [map_hash_generator-9]     from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
1666844014.7594476 [map_hash_generator-9]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/impl/implementation_singleton.py", line 32, in <module>
1666844014.7596691 [map_hash_generator-9]     rclpy_implementation = import_c_library('._rclpy_pybind11', package)
1666844014.7604921 [map_hash_generator-9]   File "/opt/ros/galactic/lib/python3.8/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
1666844014.7633743 [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/perception/object_recognition/detection/voxel_based_compare_map_filter' in container 'pointcloud_container'
1666844014.7642131 [map_hash_generator-9]     return importlib.import_module(name, package=package)
1666844014.7665758 [map_hash_generator-9]   File "/home/human/anaconda3/lib/python3.9/importlib/__init__.py", line 127, in import_module
1666844014.7667894 [map_hash_generator-9]     return _bootstrap._gcd_import(name[level:], package, level)
1666844014.7678573 [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/default_ad_api/node/localization' in container '/default_ad_api/container'
1666844014.7699301 [map_hash_generator-9] ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
1666844014.7721665 [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/autoware_api/external/start' in container '/autoware_api/external/autoware_iv_adaptor'
1666844014.7732270 [map_hash_generator-9] The C extension '/opt/ros/galactic/lib/python3.8/site-packages/_rclpy_pybind11.cpython-39-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#import-failing-without-library-present-on-the-system' for possible solutions
````

It seems your python environment is mixed at ros galactic python3.8 and anaconda python3.9.
Could you check your python environment?

hangger-258, 2022-10-27:
Thanks for your answer, actually the problem existed before I install anaconda, and I try to uninstall the anaconda and start the AWSIM-autoware again, the problem is still there.

shmpwk, 2022-10-27:
@hangger-258 
Thank you for your update.
Logging at your new log, `lidar_centerpoint` node dies at L220
```
1666865292.2733142 [ERROR] [lidar_centerpoint_node-29]: process has died [pid 10292, exit code -6, cmd '/home/human/Desktop/AWSIM-autoware/autoware/install/lidar_centerpoint/lib/lidar_centerpoint/lidar_centerpoint_node --ros-args -r __node:=lidar_centerpoint -r __ns:=/perception/object_recognition/detection/centerpoint --params-file /tmp/launch_params_7fyt8v5w --params-file /tmp/launch_params_902az41u --params-file /tmp/launch_params_rya8jghl --params-file /tmp/launch_params_l8d4cp5o --params-file /tmp/launch_params_y76047qy --params-file /tmp/launch_params_vsbzfgia --params-file /tmp/launch_params_ql82ad0x --params-file /tmp/launch_params_1vifqbki --params-file /tmp/launch_params_02clc01e --params-file /tmp/launch_params_ps6y_pam --params-file /home/human/Desktop/AWSIM-autoware/autoware/install/lidar_centerpoint/share/lidar_centerpoint/config/centerpoint_tiny.param.yaml --params-file /home/human/Desktop/AWSIM-autoware/autoware/install/lidar_centerpoint/share/lidar_centerpoint/config/detection_class_remapper.param.yaml -r ~/input/pointcloud:=/sensing/lidar/concatenated/pointcloud -r ~/output/objects:=objects'].
```

This sometimes happens when your NVIDIA GPU environment is broken.
Could you share your GPU enviroment info? 
Also, could you share the output when executing `nvidia-smi` on your console?

hangger-258, 2022-10-28:
Ubuntu version :5.15.0-52-generic #58~20.04.1-Ubuntu
Ros2: galactic
CPU 11th i9, GPU: 3070 8G
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.141.03   Driver Version: 470.141.03   CUDA Version: 11.4     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
|  0%   32C    P8     8W / 220W |    715MiB /  7979MiB |      1%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

shmpwk, 2022-10-28:
Hmm... not sure but CUDA version is one possible cause of the problem.

Currently, we use CUDA11.6. https://github.com/autowarefoundation/autoware/blob/e108b8fa9e47307b1f2573be5d0dde828235bcb0/amd64.env

In the tutorial, if you execute `./setup-dev-env.sh` without prepared CUDA environment, appropriate CUDA will be installed. 
However, if you have prepared CUDA environment, it won't.

Your CUDA11.4 is not recommend but it might work...

shmpwk, 2022-10-28:
> Additionally, The autoware can work itself in the official tutorial of autoware.
You mean you succeed the Autoware tutorial : [planning simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/) and [rosbag replay simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/)?

shmpwk, 2022-10-28:
@hangger-258 
Could you try to add `pcl_ros` in `universe/autoware.universe/sensing/pointcloud_preprocessor/CMakeLists.txt` following https://github.com/autowarefoundation/autoware.universe/pull/2101/files?

shmpwk, 2022-10-28:
This might be a solution
https://github.com/tier4/AWSIM/issues/15#issuecomment-1294609631

hangger-258, 2022-10-28:
Thank you very much! I will try it.

shmpwk, 2022-10-28:
I updated the awsim-stable branch by https://github.com/autowarefoundation/autoware/pull/2982.

shmpwk, 2022-11-07:
@hangger-258 How's it going?

mackierx111, 2022-11-20:
This issue is closed. Please open it if necessary.

closed 2022-11-20 by mackierx111