autowarefoundation/AWSIM #19: RGL Plugin error

AlonCuve, 2022-10-28:
Hi,
I want AWSIM to work on Win10.
After installing Unity 2021.1.7f and creating a project, I got an error when playing.
Please tell me what should I do?
thank you

↓↓↓Console error
RGLUnityPlugin.RGLNativeAPI.CheckErr (System.Int32 status) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNativeAPI.cs:78)
RGLUnityPlugin.RGLLidar.RaytraceAsync (UnityEngine.Matrix4x4 lidarToWorld, UnityEngine.Matrix4x4 postRaycastTransform, System.Single range) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLLidar.cs:92)
RGLUnityPlugin.LidarSensor.RequestCapture () (at Assets/RGLUnityPlugin/Scripts/LidarSensor.cs:162)
RGLUnityPlugin.LidarSensor.FixedUpdate () (at Assets/RGLUnityPlugin/Scripts/LidarSensor.cs:155)

prybicki, 2022-10-28:
Hi @AlonCuve!

The error you got means there is a problem running the LiDAR simulation on the GPU.
To help us find the problem, please provide your:
- GPU model
- GPU driver version

AlonCuve, 2022-10-31:
Hi, Thank you
GPU model:Quadro P6000
version:451.48

prybicki, 2022-10-31:
Thank you! The driver version needs to be >=460.82 on Windows. I made a PR to make it clear in the documentation. You can update your driver from [the official Nvidia webpage](https://www.nvidia.com/Download/driverResults.aspx/193171/en-us/)
Please also note that Quadro P6000 does not support hardware-accelerated raytracing so the lidar simulation may have lower performance compared to RTX GPUs.

I would be happy to learn if the driver update fixed the issue.

AlonCuve, 2022-11-01:
thank you!
I fixed it by updating the driver

closed 2022-11-01 by prybicki