autowarefoundation/AWSIM #24: Running AWSIM on AWS

mackierx111, 2022-11-01:
We are in the process of verifying how AWSIM can be run on AWS. If anyone has already tried, please let me know the instance type, etc.

screny, 2022-11-02:
ubuntu入れて実行したのですが、下記のエラーになります。

■ AWS環境
g4ad.xlarge
4vcpu 16Gibメモリ

セットアップ中に、
nvidia-smiを要求されました。

apt install ubuntu-drivers-common

Command 'nvidia-smi' not found, but can be installed with:
sudo apt install nvidia-utils-390         # version 390.154-0ubuntu0.22.04.1, or
sudo apt install nvidia-utils-450-server  # version 450.203.03-0ubuntu0.22.04.1
sudo apt install nvidia-utils-470         # version 470.141.03-0ubuntu0.22.04.1
sudo apt install nvidia-utils-470-server  # version 470.141.03-0ubuntu0.22.04.1
sudo apt install nvidia-utils-510         # version 510.85.02-0ubuntu0.22.04.1
sudo apt install nvidia-utils-510-server  # version 510.85.02-0ubuntu0.22.04.1
sudo apt install nvidia-utils-515         # version 515.76+really.515.65.01-0ubuntu0.22.04.1
sudo apt install nvidia-utils-515-server  # version 515.65.01-0ubuntu0.22.04.1
sudo apt install nvidia-utils-418-server  # version 418.226.00-0ubuntu4

とりあえず nvidia-utils-510 をインストール

sudo apt install nvidia-utils-510         # version 510.85.02-0ubuntu0.22.04.1
sudo apt install nvidia-utils-510-server  # version 510.85.02-0ubuntu0.22.04.1

ubuntu@ip-:~$ nvidia-smi 
NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. 
Make sure that the latest NVIDIA driver is installed and running.

最後は、下記のエラーでした。
ubuntu@:~/AWSIM$ /home/ubuntu/AWSIM/AWSIM.x86_64
Found path: /home/ubuntu/AWSIM/AWSIM.x86_64
Segmentation fault (core dumped)

mackierx111, 2022-11-02:
@screny 
ご報告ありがとうございます。
> NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver.
Make sure that the latest NVIDIA driver is installed and running.

NVIDIAドライバのインストールがうまくできていないようです。
また、多くの方に知見を残したいため可能であれば英語で議論させていただきたいです！

---

Thank for your report.
> NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver.
Make sure that the latest NVIDIA driver is installed and running.

It seems that the NVIDIA driver is not installed properly.
Also, I'd like to discuss this in English, if at all possible, as I'd like to leave many of you with some insight!

screny, 2022-11-02:
@mackierx111

Today I changed the instance again and tried again.
set up report detail .

AWS
setup   instance g4dn.xlarge

Using the Quick Start Demo,

root@ip-:~# vi ~/.bash_profile
root@ip-~# source ~/.bash_profile
net.core.rmem_max = 2147483647
root@ip-:~# 

sudo ubuntu-drivers autoinstall
root@ip:~# sudo ubuntu-drivers autoinstall
sudo: ubuntu-drivers: command not found

sudo apt install ubuntu-drivers-common 

sudo ubuntu-drivers autoinstall

root@:~# ubuntu-drivers devices
== /sys/devices/pci0000:00/0000:00:1e.0 ==
modalias : pci:v000010DEd00001EB8sv000010DEsd000012A2bc03sc02i00
vendor   : NVIDIA Corporation
model    : TU104GL [Tesla T4]
driver   : nvidia-driver-418-server - distro non-free
driver   : nvidia-driver-515 - distro non-free recommended
driver   : nvidia-driver-515-server - distro non-free
driver   : nvidia-driver-450-server - distro non-free
driver   : nvidia-driver-510 - third-party non-free
driver   : nvidia-driver-520 - distro non-free
driver   : nvidia-driver-470 - distro non-free
driver   : nvidia-driver-510-server - distro non-free
driver   : nvidia-driver-470-server - distro non-free
driver   : xserver-xorg-video-nouveau - distro free builtin

sudo apt install nvidia-driver-470

Vulkan driverversion=[0x75a340c0] uint=[1973633216]
CreateDirectory '/tmp/TIERIV' failed:  (current dir: /home/ubuntu/AWSIM/AWSIM_Data)
CreateDirectory '/tmp/TIERIV' failed:  (current dir: /home/ubuntu/AWSIM/AWSIM_Data)
Vulkan PSO: cache data not found [path=/vulkan_pso_cache.bin]
Caught fatal signal - signo:11 code:1 errno:0 addr:(nil)
Obtained 13 stack frames.
#0  0x007ff31b0de520 in __sigaction
#1  0x007ff31bf15419 in (Unknown)
#2  0x007ff31bf2f31f in (Unknown)
#3  0x007ff31bf1f7e7 in (Unknown)
#4  0x007ff31be9bf4e in (Unknown)
#5  0x007ff31c37cd2b in (Unknown)
#6  0x007ff31c37d091 in (Unknown)
#7  0x007ff31c22ffe2 in (Unknown)
#8  0x007ff31c23ebfc in (Unknown)
#9  0x007ff31c41d079 in PlayerMain(int, char**)
#10 0x007ff31b0c5d90 in PlayerMain(int, char**)
#11 0x007ff31b0c5e40 in __libc_init_first
#12 0x00000000201029 in __libc_start_main
Segmentation fault (core dumped)
ubuntu@:~/AWSIM$ 

ubuntu@ip-:~/AWSIM$ ./AWSIM.x86_64
Set current directory to /home/ubuntu/AWSIM
Found path: /home/ubuntu/AWSIM/AWSIM.x86_64
Segmentation fault (core dumped)
ubuntu@ip-:~/AWSIM$

mackierx111, 2022-11-02:
Is nvidia-smi installed correctly?
If installed correctly, the nvidia-smi command produces the following output (although the version may be different)
```
$ nvidia-smi 
Fri Oct 14 01:41:05 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 515.65.01    Driver Version: 515.65.01    CUDA Version: 11.7     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
| 37%   31C    P8    30W / 250W |    188MiB / 11264MiB |      3%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1151      G   /usr/lib/xorg/Xorg                133MiB |
|    0   N/A  N/A      1470      G   /usr/bin/gnome-shell               45MiB |
+-----------------------------------------------------------------------------+
```
We, too, are checking to run AWSIM on AWS. If we are successful, we will report it in this issue.

screny, 2022-11-03:
I changed the AWS instance.

g4dn.xlarge

I was fine until nvidia-smi display. 

I got an error when I ran ./AWSIM.x86_64

ubuntu@ip-172-〇〇〇:~$ nvidia-smi
Thu Nov  3 06:52:29 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 510.47.03    Driver Version: 510.47.03    CUDA Version: 11.6     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  Tesla T4            On   | 00000000:00:1E.0 Off |                    0 |
| N/A   44C    P8    16W /  70W |      0MiB / 15360MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+

ubuntu@ip-172-〇〇〇:~/AWSIM$ ./AWSIM.x86_64
Set current directory to /home/ubuntu/AWSIM
Found path: /home/ubuntu/AWSIM/AWSIM.x86_64
Segmentation fault (core dumped)

Segmentation fault (core dumped)

[image]

screny, 2022-11-04:
@mackierx111 
Does it work without a desktop in ubuntu?

mackierx111, 2022-11-06:
Maybe it works. I have run another Unity binary in the past on AWS. But I don't have a lot of knowledge about AWS. We are promoting a trial to run AWSIM on AWS in my company, so please wait for further information.

screny, 2022-11-06:
@mackierx111 
Thank you. Is there somewhere I can check the detailed error?<We are promoting a trial to run AWSIM on AWS in my company,

mackierx111, 2022-11-08:
No, not yet. sorry.

screny, 2022-11-12:
@mackierx111 

I tried setup 5 times, but it was an error.

What's wrong?

nvidia-smi is OK .

ubuntu@ip-172-31-20-:~$ cd AWSIM
ubuntu@ip-172-31-20-:~/AWSIM$ ls
AWSIM.x86_64  AWSIM_Data  LinuxPlayer_s.debug  UnityPlayer.so  UnityPlayer_s.debug
ubuntu@ip-172-31-20-:~/AWSIM$ ./AWSIM.x86_64
Set current directory to /home/ubuntu/AWSIM
Found path: /home/ubuntu/AWSIM/AWSIM.x86_64
Segmentation fault (core dumped)
ubuntu@ip-172-31-20-:~/AWSIM$

screny, 2022-11-14:
@mackierx111 

I share playerlog

  vi /home/ubuntu/.config/unity3d/TIERIV/AWSIM/Player.log

[Vulkan init] extensions: name=VK_NVX_image_view_handle, enabled=0
[Vulkan init] extensions: name=VK_NVX_multiview_per_view_attributes, enabled=0
Vulkan API version 1.1.0 (1.3.194 supported by driver, 1.1.0 requested)
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[Tesla T4] id=[1eb8]
Vulkan device type 2
Vulkan driverversion=[0x7f8bc0c0] uint=[2139865280]
Vulkan PSO: cache data not found [path=/tmp/TIERIV/AWSIM/vulkan_pso_cache.bin]
Caught fatal signal - signo:11 code:1 errno:0 addr:(nil)
Obtained 12 stack frames.
#0  0x007fa01eb8e420 in funlockfile
#1  0x007fa01f85d419 in DisplayManagerVK::Initialize()
#2  0x007fa01f87731f in GfxDeviceVK::GfxDeviceVK(MemLabelId, vk::Instance const&)
#3  0x007fa01f8677e7 in CreateVKGfxDevice()
#4  0x007fa01f7e3f4e in CreateClientGfxDevice(GfxDeviceRenderer, GfxCreateDeviceFlags)
#5  0x007fa01fcc4d2b in CreateGfxDevice(GfxDeviceRenderer, GfxCreateDeviceFlags)
#6  0x007fa01fcc5091 in InitializeGfxDevice()
#7  0x007fa01fb77fe2 in InitializeEngineGraphics(bool)
#8  0x007fa01fb86bfc in PlayerInitEngineGraphics(bool)
#9  0x007fa01fd65079 in PlayerMain(int, char**)
#10 0x007fa01e9aa083 in __libc_start_main
#11 0x00000000201029 in _start

mackierx111, 2022-11-24:
@screny 
Sorry for the late reply. Hmmm..

@dazzhe 
If you have any knowledge of running AWSIM on AWS, it would be appreciated if you could share it.

mslavescu, 2023-02-16:
The instructions here wiil help you with AWS setup:

OSSDC SIM - AutowareAuto - 4 Cars Demo for Amazon re:MARS 2022
https://github.com/OSSDC/OSSDC-SIM-Demos/discussions/2

Check also my video on how I ran the SVL simulator on 4 AWS nodes (in Windows as Unity3D it is more performant on Windows then Linux):

SVL 2021.3 simulator 4 x node cluster on AWS performance test 4 - LVMS - AV-21 - 6xCams + 3xLIDARs
https://youtu.be/roQF1ZA8z5s

closed 2023-05-23 by mackierx111