autowarefoundation/AWSIM #27: Not able to see any ros topic from AWSIM

Narendarselva, 2022-11-08:
**My PC specs:**	
OS	Ubutnu 22.04
CPU	16cores and 20thread or higher
GPU	RTX3070 
Nvidia Driver (Linux) 520.61.05.
ROS2  Humble

**Issue:**
I downloaded the pre compiled AWSIM binary from the document mentioned which is V1.0.1.
I was able to run the simulator as well. But when I tried the view the topics that are published and subscribed from AWSIM , I couldnt find any.
Note: I m using ros2 humble.
Whether we need to use different binary for using with ros2 humble ?
Any updates would be very helpful.

Thanks
Narendar Selvaraj.

mackierx111, 2022-11-08:
Currently AWSIM supports ros2-galactic.
In the near future AWSIM will support ros2-humble, but that is when Autoware supports ros2-humble.
the pull request has already been prepared and is being prepared for migration to ros2-humble.  https://github.com/tier4/AWSIM/pull/11

Narendarselva, 2022-11-08:
Hi Mack,
Thanks for the quick update.
Currently autoware supports ros2 humble. (I could see the humble branch from autoware).
And even with the "awsim-stable" branch(compiled using ros2 humble) I was able to launch the e2e simulation config successfully and able to view topics from autoware.But  I m not sure why I m not able to view topics from AWSIM. Is this becuase ros2 for unity is compiled using standalone mode ?
Will it be possible to share an updated binary image with the above mentioned ros2-humble changes , that would be really helpful.

mackierx111, 2022-11-08:
As you said, I think autoware's humble support is almost ready. So.
It might work if the user builds the environment.
But AWSIM and autoware's humble support is not officially released yet. I think we will release a new version and make it available to the public in the near future.
The current options are to wait for the release of an AWSIM version that supports the new ros2-humble or to build your own ros2-humble environment.

mackierx111, 2022-11-08:
This document is a good reference for AWSIM's Unity setup.
https://tier4.github.io/AWSIM/GettingStarted/SetupUnityProject/
However, basic Unity operating skills are required.

mackierx111, 2022-11-20:
This issue is closed. Please open it if necessary.

closed 2022-11-20 by mackierx111

LiangHuangBC, 2022-12-14:
> Currently AWSIM supports ros2-galactic. In the near future AWSIM will support ros2-humble, but that is when Autoware supports ros2-humble. the pull request has already been prepared and is being prepared for migration to ros2-humble. #11

any progress or schedule on humble ?

mackierx111, 2023-02-01:
@LiangHuangBC 
AWSIM v1.1.0 is now available. Now officially support ros2 humble!