autowarefoundation/AWSIM #32: Segmentation fault when I try to setup AWSim

Fadimee, 2022-11-16:
My PC Specs:
Ubuntu 20.04
Nvidia Driver 520

$ nvidia-smi 
Error: nvidia-smi has failed because it couldn't communicate with the nvidia driver. make sure that the latest nvidia driver is installed and running.

When I open the awsim file;
Segmentation fault (core dumped) 
I get this error.

How can I solve this problem?

xmfcx, 2022-11-16:
Can you try
```sh
sudo apt purge nvidia*
sudo ubuntu-drivers autoinstall
```
and reboot.

Then check if `nvidia-smi` works.

mackierx111, 2022-11-16:
I have a few questions.

- Is it a desktop PC?
- What is the CPU and GPU?
- Did you reboot your PC after installing the nvidia driver?

Fadimee, 2022-11-16:
> 
I followed these steps but it still gave the same error. Is there anything else I can try?

Fadimee, 2022-11-16:
> I have a few questions.
> 
> * Is it a desktop PC?
> * What is the CPU and GPU?
> * Did you reboot your PC after installing the nvidia driver?

> Can you try
> 
> ```shell
> sudo apt purge nvidia*
> sudo ubuntu-drivers autoinstall
> ```
> 
> and reboot.
> 
> Then check if `nvidia-smi` works.

I followed the steps you said. It still gave the same error. Is there anything else I can try?

Fadimee, 2022-11-16:
> Can you try
> 
> ```shell
> sudo apt purge nvidia*
> sudo ubuntu-drivers autoinstall
> ```
> 
> and reboot.
> 
> Then check if `nvidia-smi` works.

I followed the steps you said. It still gave the same error. Is there anything else I can try?

Fadimee, 2022-11-16:
> I have a few questions.
> 
> * Is it a desktop PC?
> * What is the CPU and GPU?
> * Did you reboot your PC after installing the nvidia driver?

> * Yes, it is a desktop.
> * intel(r) xeon(r) w-2123 cpu 
> * Yes, I reboot my PC after installing the Nvidia driver.

mackierx111, 2022-11-16:
@Fadimee 
Do you have a GPU?

harishkumarbalaji, 2022-11-17:
I am also facing the same Issue. `nvidia-smi` works for me and here is the output,

```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 520.61.05    Driver Version: 520.61.05    CUDA Version: 11.8     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  Quadro T1000        On   | 00000000:01:00.0  On |                  N/A |
| N/A   48C    P8     5W /  N/A |    103MiB /  4096MiB |     20%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1525      G   /usr/lib/xorg/Xorg                 95MiB |
+-----------------------------------------------------------------------------+
```
But When I run `./AWSIM.x86_64`, I am getting a blank screen for some time, and then it is crashing throwing,
`[1]    63718 segmentation fault (core dumped)  ./AWSIM.x86_64`

CihatAltiparmak, 2022-11-17:
> I am also facing the same Issue. `nvidia-smi` works for me and here is the output,
>
> ```
> +-----------------------------------------------------------------------------+
> | NVIDIA-SMI 520.61.05 [...]

Hey, @harishkumarbalaji , did you source ros2 before AWSIM simulation ? If you source ros2 before AWSIM simulation, you shouldn't source ros2, i guess ros2 is already sourced when AWSIM is executed. I tried like this and i run simulation. 

Firstly, open terminal that don't source any ros distro, then in this terminal, run simulation. I hope this solves your issue.  Finally, you can look at ~/config/unity3d/TIER/AWSIM/Player.log, maybe other ones can understand cause of your issue.

harishkumarb-tw, 2022-11-18:
I tried launching the AWSIM without sourcing ROS2 Distro. Still, I am facing the same,
[image]
Also, the last lines of `~/.config/unity3d/TIERIV/AWSIM/Player.log` are,
```
=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

Caught fatal signal - signo:11 code:1 errno:0 addr:0xb0
Obtained 16 stack frames.
#0  0x007f4d93d63420 in funlockfile
#1  0x007f4c6928fcef in (Unknown)
#2  0x007f4c6926fd6b in (Unknown)
#3  0x007f4c6ac4991f in __glx_Main
#4  0x007f4d94a96035 in vk::AllocateImage(VkDevice_T*, VkPhysicalDeviceMemoryProperties const&, vk::MemoryAllocator&, VkImage_T*, unsigned int, unsigned long)
#5  0x007f4d94ad14a7 in vk::ImageManager::PrepareNewImage(TextureDimension, int, VkExtent3D const&, VkFormat, int, unsigned int, VkSampleCountFlagBits, VkFormat, bool, vk::ImageMemory, bool, bool, unsigned int)
#6  0x007f4d94ad34e0 in vk::ImageManager::CreateImage(TextureDimension, int, VkExtent3D const&, GraphicsFormat, int, unsigned int, VkSampleCountFlagBits, vk::ImageMemory)
#7  0x007f4d94a9b159 in vk::Texture::Create(vk::CommandBuffer*, TextureID, TextureDimension, int, VkExtent3D const&, GraphicsFormat, int, bool, VkSampleCountFlagBits, vk::ImageMemory)
#8  0x007f4d94ac7769 in vk::RenderSurface::CreateColorSurfaceImpl(vk::RenderSurface*, vk::CommandBuffer*, vk::ImageManager&, GraphicsFormat, VkExtent3D, int, bool, VkImage_T*)
#9  0x007f4d94ac754a in vk::RenderSurface::CreateColorSurfacePlatform(vk::RenderSurface*, vk::CommandBuffer*, vk::ImageManager&, GraphicsFormat, bool, VkImage_T*)
#10 0x007f4d955620fb in GfxDeviceWorker::RunCommand(ThreadedStreamBuffer&)
#11 0x007f4d95569ccf in GfxDeviceWorker::RunExt(ThreadedStreamBuffer&)
#12 0x007f4d95560275 in GfxDeviceWorker::RunGfxDeviceWorker(void*)
#13 0x007f4d94e5049d in Thread::RunThreadWrapper(void*)
#14 0x007f4d93d57609 in start_thread
#15 0x007f4d93c7c133 in clone
```

harishkumarbalaji, 2022-11-18:
@CihatAltiparmak  I tried launching the AWSIM without sourcing ROS2 Distro. Still, I am facing the same,
[image]
Also, the last lines of `~/.config/unity3d/TIERIV/AWSIM/Player.log` are,
```
=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

Caught fatal signal - signo:11 code:1 errno:0 addr:0xb0
Obtained 16 stack frames.
#0  0x007f4d93d63420 in funlockfile
#1  0x007f4c6928fcef in (Unknown)
#2  0x007f4c6926fd6b in (Unknown)
#3  0x007f4c6ac4991f in __glx_Main
#4  0x007f4d94a96035 in vk::AllocateImage(VkDevice_T*, VkPhysicalDeviceMemoryProperties const&, vk::MemoryAllocator&, VkImage_T*, unsigned int, unsigned long)
#5  0x007f4d94ad14a7 in vk::ImageManager::PrepareNewImage(TextureDimension, int, VkExtent3D const&, VkFormat, int, unsigned int, VkSampleCountFlagBits, VkFormat, bool, vk::ImageMemory, bool, bool, unsigned int)
#6  0x007f4d94ad34e0 in vk::ImageManager::CreateImage(TextureDimension, int, VkExtent3D const&, GraphicsFormat, int, unsigned int, VkSampleCountFlagBits, vk::ImageMemory)
#7  0x007f4d94a9b159 in vk::Texture::Create(vk::CommandBuffer*, TextureID, TextureDimension, int, VkExtent3D const&, GraphicsFormat, int, bool, VkSampleCountFlagBits, vk::ImageMemory)
#8  0x007f4d94ac7769 in vk::RenderSurface::CreateColorSurfaceImpl(vk::RenderSurface*, vk::CommandBuffer*, vk::ImageManager&, GraphicsFormat, VkExtent3D, int, bool, VkImage_T*)
#9  0x007f4d94ac754a in vk::RenderSurface::CreateColorSurfacePlatform(vk::RenderSurface*, vk::CommandBuffer*, vk::ImageManager&, GraphicsFormat, bool, VkImage_T*)
#10 0x007f4d955620fb in GfxDeviceWorker::RunCommand(ThreadedStreamBuffer&)
#11 0x007f4d95569ccf in GfxDeviceWorker::RunExt(ThreadedStreamBuffer&)
#12 0x007f4d95560275 in GfxDeviceWorker::RunGfxDeviceWorker(void*)
#13 0x007f4d94e5049d in Thread::RunThreadWrapper(void*)
#14 0x007f4d93d57609 in start_thread
#15 0x007f4d93c7c133 in clone
```

tleyden, 2022-11-18:
I also got a segmentation fault + core dump when running `AWSim` in a docker container, but it worked fine on the linux host.  I added some more details here: https://github.com/autowarefoundation/autoware/discussions/2991, including output from `vulkaninfo` in both the host and container.

mackierx111, 2022-11-20:
Thanks for sharing.
The cause of the Segmentation fault error is most likely the Nvidia driver or vulkan API.

By the way, the specs of the desktop PC on which I ran the AWSIM demo are currently as follows

specs 1
- CPU : 9900k
- GPU : RTX2080Ti
- OS : Ubutnu 20.04

specs 2
- GPU : 5950X
- GPU : RTX3090
- OS : Ubuntu 20.04

I have yet to run it on AWS or other instances or Docker.

mackierx111, 2022-12-07:
This issue is closed. Please open it if necessary.

closed 2022-12-07 by mackierx111

sdudiak-husarion, 2023-02-14:
I am getting an the same error as previously described. After installing everything according to [Quick Start Demo](https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/), an AWSIM window opens for a short time and after that i get the segmentation fault error. Here is a part of output from my `~/.config/unity3d/TIERV/AWSIM/Player.log`:
```bash
=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

Caught fatal signal - signo:11 code:1 errno:0 addr:0xf0
Obtained 16 stack frames.
#0  0x007fb8344da520 in __sigaction
#1  0x007fb831a0d2fc in (Unknown)
#2  0x007fb8319ebff2 in (Unknown)
#3  0x007fb8330125f0 in __glx_Main
#4  0x007fb835346035 in vk::AllocateImage(VkDevice_T*, VkPhysicalDeviceMemoryProperties const&, vk::MemoryAllocator&, VkImage_T*, unsigned int, unsigned long)
#5  0x007fb8353814a7 in vk::ImageManager::PrepareNewImage(TextureDimension, int, VkExtent3D const&, VkFormat, int, unsigned int, VkSampleCountFlagBits, VkFormat, bool, vk::ImageMemory, bool, bool, unsigned int)
#6  0x007fb8353834e0 in vk::ImageManager::CreateImage(TextureDimension, int, VkExtent3D const&, GraphicsFormat, int, unsigned int, VkSampleCountFlagBits, vk::ImageMemory)
#7  0x007fb83534b159 in vk::Texture::Create(vk::CommandBuffer*, TextureID, TextureDimension, int, VkExtent3D const&, GraphicsFormat, int, bool, VkSampleCountFlagBits, vk::ImageMemory)
#8  0x007fb835377769 in vk::RenderSurface::CreateColorSurfaceImpl(vk::RenderSurface*, vk::CommandBuffer*, vk::ImageManager&, GraphicsFormat, VkExtent3D, int, bool, VkImage_T*)
#9  0x007fb83537754a in vk::RenderSurface::CreateColorSurfacePlatform(vk::RenderSurface*, vk::CommandBuffer*, vk::ImageManager&, GraphicsFormat, bool, VkImage_T*)
#10 0x007fb835e120fb in GfxDeviceWorker::RunCommand(ThreadedStreamBuffer&)
#11 0x007fb835e19ccf in GfxDeviceWorker::RunExt(ThreadedStreamBuffer&)
#12 0x007fb835e10275 in GfxDeviceWorker::RunGfxDeviceWorker(void*)
#13 0x007fb83570049d in Thread::RunThreadWrapper(void*)
#14 0x007fb83452cb43 in pthread_condattr_setpshared
#15 0x007fb8345bea00 in __xmknodat

```
If necessary I can provide the entire file.

My computer specification:
```bash
OS: Ubuntu 22.04.1 LTS x86_64 
CPU: Intel i5-9300HF (8) @ 4.100GHz
GPU: NVIDIA GeForce GTX 1650 Mobile / Max-Q
```

My output of `nvidia-smi` : 
```bash
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.60.13    Driver Version: 525.60.13    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
| N/A   48C    P0    13W /  N/A |    464MiB /  4096MiB |     27%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2020      G   /usr/lib/xorg/Xorg                253MiB |
|    0   N/A  N/A      2220      G   /usr/bin/gnome-shell               55MiB |
|    0   N/A  N/A      4014      G   ...358965108654495738,131072      122MiB |
+-----------------------------------------------------------------------------+
```
Is this issue caused by using GTX1650 instead of some RTX model? If so, is there any way to launch it on GTX1650?