autowarefoundation/AWSIM #35: Setting up Kashiwanoha Map

yuasabe, 2022-11-19:
I am interested in setting up the Kashiwanoha Map environment (https://github.com/tier4/AWSIM/tree/feature/kashiwanoha-map) in AWSIM, so that Autoware and possibly NPC Random Traffic generation also works. A couple of questions I wanted to ask:

- What do I need to do in the Unity Editor to set up the Kashiwanoha map? 
  - I believe that `mgrsOffsetPosition ` needs to be set properly. What is the `mgrsOffsetPosition ` for the Kashiwanoha map?
- Which Lanelet2 and PCD map should I use in Autoware?
- Any other settings I must change or add?

mackierx111, 2022-11-20:
@yuasabe 
I see, you want to connect AWSIM and Autoware in Kashiwanoha-map.

> I believe that mgrsOffsetPosition needs to be set properly. What is the mgrsOffsetPosition for the Kashiwanoha map?

yes you are correct.
But, without VectorMap, it would be difficult. Since VectorMap is not yet available to the public, I will check if it is possible to publish it from our company.

mackierx111, 2022-11-24:
@yuasabe 
Please check out .pcd and .osm added to `feature/kashiwanoha-map` latest.
https://github.com/tier4/AWSIM/commit/8795936b97511fb73f7d33dee732c0ba22876097
Please wait as we have not yet completed setting up the MGRS offset position.

closed 2023-01-30 by mackierx111