autowarefoundation/AWSIM #36: UnityHub.Appimage doesn't run on Ubuntu 22.04

xmfcx, 2022-11-23:
After downloading `UnityHub.Appimage`, running it didn't do anything.

I was able to run it with `./UnityHub.AppImage --disable-gpu-sandbox` command following advice at https://forum.unity.com/threads/running-unity-on-ubuntu-22-04.1284083/

We might need to update the [docs here](https://tier4.github.io/AWSIM/GettingStarted/SetupUnityProject/#2-install-unity-202117f1).

Did anyone else face this issue?

xmfcx, 2022-11-24:
```

mackierx111, 2022-11-24:
@xmfcx 
thank you!
We will be supporting AWSIM to Ubuntu 22.04 and ros2-humble in the future, so I will refer to this issue. 
I will also inform Unity of this issue.

mackierx111, 2023-01-30:
Supported ros2 humble since v1.1.0
https://github.com/tier4/AWSIM/releases/tag/v1.1.0

closed 2023-01-30 by mackierx111