autowarefoundation/AWSIM #38: Map does not Load in Rviz

marcusvinicius178, 2022-11-24:
Hi I was following the QuickDemo tutorial https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/
I am running in NUC. Ubuntu 20.04. I have ROS2 Galactic installed, Nivida-Driver 520 Cuda 11.8
I was able to build autoware repo and checkout into the aw-stable branch....

I believe that I am setting the argument uncorrectly or is there other issue ?

[image]

As you can see Rviz opens with a black window instead the map

I have lodaded the maps inside the **autoware/src** folder and launched the launch file as described below:

`ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit map_path:=/home/rota/autoware/src/nishishinjuku_autoware_map`

This is the path: /home/rota/autoware/src/nishishinjuku_autoware_map
where the pcd and lanelet2 files from nishishinjuku map are

[image]

I imagined that I needed a name to the **map_path:=** argument? Is something missing?

Thanks in advance

marcusvinicius178, 2022-11-24:
Solved the issue closing the terminal with launch file and relaunching again. 
Actually I needed to kiill all process related with rviz using:

pkill -9 -f rviz

It worked as it can be seen in my video
https://drive.google.com/file/d/1vwceUrZDRwxx-4BxgPJ7SAundk9xbZvv/view?usp=share_link

Very nice simulator!

closed 2022-11-24 by marcusvinicius178

mackierx111, 2022-11-25:
Glad to hear you got it sorted out!
I watched the video. The initial position of the vehicle seems to be different, so if you can initialpose well, you can run Autoware smoothly.

marcusvinicius178, 2022-12-01:
Hi, Mr. @mackierx111 I am sorry to reuse this channel. I have a quick question ( I have also sent an e-mail to the tier4 team recently about it).

We will do our **first test** next month in our Truck using Autoware Universe (obviously in a test field, not a complex environment to use NDT). But we will **use just GPS (RTK) data**, which will be the only source for **localization** and planning modules. Is it enough? Does Autoware **can receive GPS data**, or must I work around it to enable it? 

I would be happy if you could provide me with links and info.