autowarefoundation/AWSIM #41: Segmentation fault (core dumped) occurred. with vulkan pso not found info.

JohnnyHan, 2022-11-28:
1）run the command beblow
workspace/autoware/AWSIM_v1.0.1# ./AWSIM/AWSIM.x86_64

2) and get the error info 

Set current directory to /workspace/autoware/AWSIM_v1.0.1
Found path: /workspace/autoware/AWSIM_v1.0.1/AWSIM/AWSIM.x86_64
Segmentation fault (core dumped)

3）cat ~/.config/unity3
d/TIERIV/AWSIM/Player.log 

[Vulkan init] extensions: name=VK_NVX_multiview_per_view_attributes, enabled=0
Vulkan API version 1.1.0 (1.3.205 supported by driver, 1.1.0 requested)
Vulkan vendor=[NVIDIA] id=[10de]
Vulkan renderer=[NVIDIA GeForce RTX 3090] id=[2204]
Vulkan device type 2
Vulkan driverversion=[0x80d04040] uint=[-2133835712]
Vulkan PSO: cache data not found [path=/tmp/TIERIV/AWSIM/vulkan_pso_cache.bin]
Caught fatal signal - signo:11 code:1 errno:0 addr:0x13

4） sys info
cpu: Intel(R) Xeon(R) Gold 6226R CPU @ 2.90GHz
gpu: Nvidia 3090

JohnnyHan, 2022-11-28:
It's successfully running in my service with ubuntu20.04 system.
but still Segmentation fault (core dumped) in the container with ubuntu20.04

mackierx111, 2022-11-29:
Thanks for the report. I have yet to run it in a container too.
An issue is standing, but it is not a priority in the current situation.
https://github.com/tier4/AWSIM/issues/7

closed 2022-12-22 by mackierx111