autowarefoundation/AWSIM #42: RobotecGPULidar error

AlonCuve, 2022-12-01:
Hi,
Please help me solve my problem.
Unity console error ↓↓↓↓↓↓↓↓↓↓↓↓↓↓

・DllNotFoundException: RobotecGPULidar
RGLUnityPlugin.RGLNativeAPI.NodeRaysFromMat3x4f (System.IntPtr& node, UnityEngine.Matrix4x4[] rays) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNativeAPI.cs:165)
RGLUnityPlugin.RGLNodeSequence.AddNodeRaysFromMat3x4f (System.String identifier, UnityEngine.Matrix4x4[] rays) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNodeSequence.cs:73)
RGLUnityPlugin.LidarSensor.Awake () (at Assets/RGLUnityPlugin/Scripts/LidarSensor.cs:82)

・DllNotFoundException: RobotecGPULidar
RGLUnityPlugin.RGLNativeAPI.CheckVersion () (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNativeAPI.cs:107)
RGLUnityPlugin.SceneManager.Awake () (at Assets/RGLUnityPlugin/Scripts/SceneManager.cs:75)

・DllNotFoundException: RobotecGPULidar
RGLUnityPlugin.RGLNativeAPI.NodePointsTransform (System.IntPtr& node, UnityEngine.Matrix4x4 transform) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNativeAPI.cs:200)
RGLUnityPlugin.RGLNodeSequence.AddNodePointsTransform (System.String identifier, UnityEngine.Matrix4x4 transform) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNodeSequence.cs:117)
AWSIM.RglLidarPublisher.Start () (at Assets/AWSIM/Scripts/Sensors/LiDAR/RglLidarPublisher.cs:67)

Windows10
GPU model:Quadro P6000
version:517.40

prybicki, 2022-12-01:
Thanks for the report @AlonCuve!

I've seen `DllNotFoundException` once reported by @mackierx111, AFAIR upgrading drivers helped in @mackierx111's case. However, since you have a sufficient driver version I think it is a different underlying issue, since `DllNotFoundException` may be caused by a large class of issues.

Please let me ask you about the changes you have introduced to your environment since fixing #19 and some additional questions:
- Are you using AWSIM in Unity Editor or as a built binary?
- Could you specify the exact version (e.g. commit SHA1 or link to the pre-built binary) of AWSIM? It would be very helpful if you could find (`git bisect`) the first version that does not work for you.

In the meantime, I will work to add additional handling of this type of error, and then I will ask you to re-run it which will provide more details helpful for debugging.

AlonCuve, 2022-12-02:
Thank you!

・Unity Editor
・AWSIM version v1.0.1  https://github.com/tier4/AWSIM.git

msz-rai, 2022-12-02:
Hi @AlonCuve,
Thank you for the details.

I have tried to handle this type of error and get more information about missing dll, but I haven't been able to get anything concrete. My intuition tells me that some of dependencies of RobotecGPULidar were not found. The easiest way to check shared dependencies is to use external tools, such as [ldd](https://man7.org/linux/man-pages/man1/ldd.1.html).

Please let me ask you about the output from the `ldd` command  on `RobotecGPULidar.dll` file located in `[your-AWSIM-project-dir]/Assets/RGLUnityPlugin/Plugins/Windows/x86_64/`. To do this, you need to install [Cygwin](https://www.cygwin.com/) on your Windows and then start Cygwin Terminal or use [git bash terminal on Windows](https://gitforwindows.org/index.html).

Also, you can check the newest version of RGL (v0.11.2), which is provided in PR [43](https://github.com/tier4/AWSIM/pull/43). A significant change compared to the previous version is CUDA Runtime linked statically.

prybicki, 2022-12-19:
@AlonCuve Did you manage to fix the error?

AlonCuve, 2022-12-26:
I can't,
It's difficult for me. Isn't there any other method?

AlonCuve, 2022-12-26:
Hi,
git clone the AWSIM repository again
I solved the problem.

thank you!

closed 2022-12-26 by AlonCuve