autowarefoundation/AWSIM #59: GnuTLS recv error

yumianhuli2, 2023-01-03:
github couldn't download it for this reason below：

[image]

Manual git clone should be OK. Could you plz tell me where to put the content after git clone?

shmpwk, 2023-01-05:
> Could you plz tell me where to put the content after git clone?
 
Your blue sentenses tell the location of the vcs imported codes.

yumianhuli2, 2023-01-20:
autoware.repos has some versions/branch is not currently on a branch.e.g. url: https://github.com/tier4/pointcloud_to_laserscan.git
    version: 948a4fca35dcb03c6c8fbfa610a686f7c919fe0b

if I use git pull command, terminal shows:
Please specify which branch you want to merge with.
See git-pull(1) for details.

    git pull <remote> <branch>