autowarefoundation/AWSIM #64: [Question]AWSIM provide interface data like Lidar /Radar(sensor data) to other application or program bin's

anilbommareddy, 2023-01-13:
is AWSIM provide interface data like lidar/radar(sensor) data to other application or Hardware.
For an example I have an application(non ROS) which calculate the position of a tag and i need the get this positions of data(x,y,z) from AWSIM and the calculated position tag need to display on AWSIM GUI.

shmpwk, 2023-01-20:
@anilbommareddy 
Currently we don't support it.
 
@mackierx111  Do you have some comments?

mackierx111, 2023-02-14:
@anilbommareddy 
Sorry for the late reply.
For now, the officially supported data format is ROS2 msg, which is used by Autoware. However, it is possible to output the data in any data format with user customization.

gabrik, 2023-02-15:
> @anilbommareddy Sorry for the late reply. For now, the officially supported data format is ROS2 msg, which is used by Autoware. However, it is possible to output the data in any data format with user customization.

Hi, is there any example of how to extract data from the simulator without using ROS2 for communication?

anilbommareddy, 2023-02-15:
> @anilbommareddy Sorry for the late reply. For now, the officially supported data format is ROS2 msg, which is used by Autoware. However, it is possible to output the data in any data format with user customization.

@mackierx111  thank you for the response and update. I would like to know the data format is ROS2 msg you mean like Publisher and Subscriber node. Is any example to output & input the data format with user customization,?

mackierx111, 2023-02-28:
@gabrik @anilbommareddy 
No exampleple, but can be easily explained. For example, let's say you want to use a different method for GNSS sensor output in AWSIM instead of ROS. `GnssSensor.prefab` has the scripts` GnssSensor.cs` and `GnssRos2Publisher.cs` attached.
First, remove the GnssRos2Publisher component from the prefab. And `GnssSensor.cs` has a delegate named `OnOutputData` as a member. This delegate can register methods in which `GnssSensor.OutputData` is an argument. In other words, data acquired from `OutputData` can be freely converted, output, etc.

The implementation for publishing in the default ros2 is as follows
https://github.com/tier4/AWSIM/blob/main/Assets/AWSIM/Scripts/Sensors/Gnss/GnssRos2Publisher.cs#L46

closed 2023-05-09 by mackierx111