autowarefoundation/AWSIM #71: Assets/Ros2ForUnity/Plugins/Windows seems to be missing (v1.1.0 and main)

genarks, 2023-01-19:
Assets/Ros2ForUnity/Plugins/Windows seems to be missing.

When the simulator is running on Windows, the following log will appear in the Unity Console.
[image]

(I am new to Autoware. Sorry if I am misguided.)

shmpwk, 2023-01-20:
@genarks 
Unfortunately, the latest v1.1.0 does not support windows now as described at https://github.com/tier4/AWSIM/releases/tag/v1.1.0 .

Please use v1.0.2 https://github.com/tier4/AWSIM/tree/v1.0.2 for now.

genarks, 2023-01-20:
Sorry, I did not read that description in v1.1.0.

I will try to use v1.0.2.

Thank you very much.

closed 2023-01-20 by genarks