autowarefoundation/AWSIM #73: rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO

joseph16388, 2023-01-20:
I fall ollowed doc and got this error message: 

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
behavior_path_planner: Cannot locate rosdep definition for [behaviortree_cpp_v3]
tensorrt_yolox: Cannot locate rosdep definition for [tensorrt_cmake_module]
fault_injection: Cannot locate rosdep definition for [diagnostic_updater]
planning_error_monitor: Cannot locate rosdep definition for [diagnostic_updater]
trtexec_vendor: Cannot locate rosdep definition for [tensorrt_cmake_module]
grid_map_ros: Cannot locate rosdep definition for [nav2_msgs]
image_projection_based_fusion: Cannot locate rosdep definition for [pcl_ros]
radar_threshold_filter: Cannot locate rosdep definition for [radar_msgs]
sample_sensor_kit_launch: Cannot locate rosdep definition for [usb_cam]
pacmod_interface: Cannot locate rosdep definition for [diagnostic_updater]
accel_brake_map_calibrator: Cannot locate rosdep definition for [diagnostic_updater]
tier4_api_msgs: Cannot locate rosdep definition for [geographic_msgs]
lidar_apollo_segmentation_tvm: Cannot locate rosdep definition for [tvm_vendor]
grid_map_cv: Cannot locate rosdep definition for [filters]
sample_vehicle_description: Cannot locate rosdep definition for [xacro]
tier4_autoware_api_launch: Cannot locate rosdep definition for [topic_tools]
lidar_centerpoint: Cannot locate rosdep definition for [pcl_ros]
external_cmd_selector: Cannot locate rosdep definition for [diagnostic_updater]
tvm_utility: Cannot locate rosdep definition for [tvm_vendor]
image_diagnostics: Cannot locate rosdep definition for [diagnostic_updater]
lidar_apollo_segmentation_tvm_nodes: Cannot locate rosdep definition for [ros_testing]
system_monitor: Cannot locate rosdep definition for [diagnostic_updater]
obstacle_stop_planner: Cannot locate rosdep definition for [pcl_ros]
trajectory_follower: Cannot locate rosdep definition for [diagnostic_updater]
topic_state_monitor: Cannot locate rosdep definition for [diagnostic_updater]
grid_map_costmap_2d: Cannot locate rosdep definition for [nav2_costmap_2d]
velodyne_driver: Cannot locate rosdep definition for [diagnostic_updater]
tensorrt_common: Cannot locate rosdep definition for [tensorrt_cmake_module]
pointcloud_preprocessor: Cannot locate rosdep definition for [point_cloud_msg_wrapper]
radar_static_pointcloud_filter: Cannot locate rosdep definition for [radar_msgs]
grid_map_filters: Cannot locate rosdep definition for [filters]
lane_departure_checker: Cannot locate rosdep definition for [diagnostic_updater]
lanelet2_extension: Cannot locate rosdep definition for [lanelet2_validation]
grid_map_demos: Cannot locate rosdep definition for [octomap_msgs]
awsim_sensor_kit_launch: Cannot locate rosdep definition for [usb_cam]
scenario_selector: Cannot locate rosdep definition for [topic_tools]
radar_tracks_msgs_converter: Cannot locate rosdep definition for [radar_msgs]
vehicle_cmd_gate: Cannot locate rosdep definition for [diagnostic_updater]
ground_segmentation: Cannot locate rosdep definition for [pcl_ros]
obstacle_velocity_limiter: Cannot locate rosdep definition for [pcl_ros]
grid_map_octomap: Cannot locate rosdep definition for [octomap]
autoware_point_types: Cannot locate rosdep definition for [point_cloud_msg_wrapper]
obstacle_collision_checker: Cannot locate rosdep definition for [pcl_ros]
autoware_auto_geometry: Cannot locate rosdep definition for [osrf_testing_tools_cpp]
map_loader: Cannot locate rosdep definition for [pcl_ros]
web_controller: Cannot locate rosdep definition for [rosbridge_server]
lanelet2_map_preprocessor: Cannot locate rosdep definition for [pcl_ros]
velodyne_monitor: Cannot locate rosdep definition for [diagnostic_updater]
probabilistic_occupancy_grid_map: Cannot locate rosdep definition for [pcl_ros]
tamagawa_imu_driver: Cannot locate rosdep definition for [can_msgs]
steer_offset_estimator: Cannot locate rosdep definition for [diagnostic_updater]
dummy_diag_publisher: Cannot locate rosdep definition for [diagnostic_updater]
bluetooth_monitor: Cannot locate rosdep definition for [diagnostic_updater]
lidar_centerpoint_tvm: Cannot locate rosdep definition for [tvm_vendor]
external_cmd_converter: Cannot locate rosdep definition for [diagnostic_updater]
front_vehicle_velocity_estimator: Cannot locate rosdep definition for [pcl_ros]
costmap_generator: Cannot locate rosdep definition for [pcl_ros]
tier4_vehicle_launch: Cannot locate rosdep definition for [xacro]
autoware_testing: Cannot locate rosdep definition for [ros_testing]
localization_error_monitor: Cannot locate rosdep definition for [diagnostic_updater]
awapi_awiv_adapter: Cannot locate rosdep definition for [topic_tools]
occupancy_grid_map_outlier_filter: Cannot locate rosdep definition for [pcl_ros]
external_velocity_limit_selector: Cannot locate rosdep definition for [topic_tools]
system_error_monitor: Cannot locate rosdep definition for [rqt_robot_monitor]
osqp_interface: Cannot locate rosdep definition for [osqp_vendor]
tier4_localization_launch: Cannot locate rosdep definition for [topic_tools]
velodyne_pointcloud: Cannot locate rosdep definition for [pcl_ros]
velodyne_description: Cannot locate rosdep definition for [xacro]

plz help , thank you!

shmpwk, 2023-01-20:
Did you solve your previous error? 
https://github.com/tier4/AWSIM/issues/58

Could you share your environment information?
Also did you source installing the Autoware as [the tutorial](https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/) says?

shmpwk, 2023-01-21:
@joseph16388 
For Ubuntu 20.04, please reffer to v1.0.2  the https://github.com/tier4/AWSIM/releases/tag/v1.0.2.
From last week, the latest main branch is for Ubuntu22.04.

Also, could you try 
```
 rosdep update --include-eol-distros
```
before `rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO` ?

(related: https://github.com/tier4/AWSIM/pull/68/files)

joseph16388, 2023-01-22:
ok , and  then I use command rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
I got
given path 'src' does not exist

shmpwk, 2023-01-22:
You can execute that at autoware repository which has sec directory.

joseph16388, 2023-01-23:
> You can execute that on `autoware` repository which has `src` directory.

Yes,done from autoware repository which has src directory!
And I run this command colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-w" now
[ 11%] Performing download step (git clone) for 'tensorrt-populate'
Cloning into 'tensorrt-src'...
fatal: unable to access 'https://github.com/NVIDIA/TensorRT/': GnuTLS recv error (-110): The TLS connection was non-properly terminated.

Could you plz tell me where can I change the context of https://github.com/NVIDIA/TensorRT/ to git://github.com//NVIDIA/TensorRT/

thank you!

shmpwk, 2023-01-23:
NVIDIA related libraries should be automatically configured with
```
./setup-dev-env.sh
```
described at https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/, Launching Autoware, 4. Configure the environment.

shmpwk, 2023-01-25:
Could you try 
```
sudo apt update
rosdep update --include-eol-distros
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
```
again?

joseph16388, 2023-01-27:
> @joseph16388 Could you try
> 
> ```
> sudo apt update
> rosdep update --include-eol-distros
> rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
> ```
> 
> again?

 Hello,I tried again
this time it shows:
[image]

shmpwk, 2023-01-27:
@joseph16388 

Could you try this?
```
rm install/share/trtexec_vendor/ build/trtexec_vendor/
sudo apt update
rosdep update --include-eol-distros
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
```

Also could you let me confirm that you use Autoware `galactic` branch by 
```
git branch
```
under autoware repository.

shmpwk, 2023-01-27:
Did you install CUDA by `./setup-dev-env.sh` ?

joseph16388, 2023-01-27:
> Did you install CUDA and its related libraries by `./setup-dev-env.sh` ?

Second trying passed. Thank you!

joseph16388, 2023-01-27:
> @joseph16388
> 
> Could you try this?
> 
> ```
> rm install/share/trtexec_vendor/ build/trtexec_vendor/
> sudo apt update
> rosdep update --include-eol-distros
> rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
> ```
> 
> Also could you confirm that you use Autoware `galactic` branch by
> 
> ```
> git branch
> ```
> 
> under autoware repository.

Second trying passed. Thank you!

shmpwk, 2023-01-27:
great!

closed 2023-01-27 by shmpwk