autowarefoundation/AWSIM #76: Vehicle init pose: failed to get transform from map to base_link

Fadimee, 2023-01-24:
Hello
I am trying to run "Quick Start Demo" on AWSIM. I am encountering a problem about vehicle pose. 
Autoware can not find init pose of vehilce and when I try to do it manually by using "2D Pose Estimate" still it doesnt work. Vehicle's init pose on Autoware is very far from map origin. 

I shared screenshots and logs below;

Logs;
`[motion_velocity_smoother-58] [WARN] [1674567417.778344619] [planning.scenario_planning.motion_velocity_smoother]: failed to get transform from map to base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[motion_velocity_smoother-58] [INFO] [1674567417.778411542] [planning.scenario_planning.motion_velocity_smoother]: waiting for self pose...
[component_container-64] [WARN] [1674567418.757881289] [control.trajectory_follower.lane_departure_checker_node]: failed to get transform from map to base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[component_container-64] [INFO] [1674567418.757951568] [control.trajectory_follower.lane_departure_checker_node]: waiting for self pose...
[component_container-62] [INFO] [1674567421.019612140] [planning.scenario_planning.parking.costmap_generator]: waiting for initial pose...
[component_container_mt-61] [WARN] [1674567422.255320688] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: failed to get transform from map to base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[component_container_mt-61] [INFO] [1674567422.255387806] [planning.scenario_planning.lane_driving.motion_planning.obstacle_avoidance_planner]: waiting for self pose...
`
Screenshots;
[image]
[image]
[image]

How can I solve this problem?

shmpwk, 2023-01-25:
@Fadimee 
Could you share your environment information?

Fadimee, 2023-01-26:
> @Fadimee Could you share your environment information?

My environment is Japan Tokyo Nishishinjuku in the Quick Start Demo, that can be seen in the given link below:

https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/

shmpwk, 2023-01-26:
What environment mean is Ubuntu, ROS version and etc.

Fadimee, 2023-01-27:
I misunderstood
Ubuntu 22.04 
Ros2 Humble 
AWSIM v1.0.1

shmpwk, 2023-01-27:
@Fadimee 

AWSIM v1.0.1 is for Ubuntu 20, ROS2 Galactic version.
Could you reffer to AWSIM v1.1.0 or the latest version on main branch instead? 
Note that if you build Autoware again, I recommend you to once remove build and install directory under autoware directory.

Fadimee, 2023-01-30:
Hello ;
Thanks for your advice. But ı already tried it before create the issue. I tried after your advice again but the result was same.

I want to give more details about the issue;
[image]
[image]
[image]

shmpwk, 2023-01-31:
Could you reffer to this discussion?
https://github.com/orgs/autowarefoundation/discussions/3236

Could you share your log of Autoware from the start to error part?

Fadimee, 2023-02-01:
@shmpwk 
I tried your suggestions on discussion you referred. Still same result. Here some other details:

- Planning simulation and rosbag replay simulation tutorial works correctly.
- When I check gpu with nvidia-smi I can see AWSIM use GPU. You can see in attachments.
- Waiting for initialize of AWSIM or Autoware doesnt change result.
- There are not cuda initialize error in Autoware output.
- There are a new error on autoware output after reinstall autoware: 
 '/localization/initialize: status code 1 'The vehicle is not stopped.'
- You can find other details you asked on [discussion 3236](https://github.com/orgs/autowarefoundation/discussions/3236) in attachments

[Autoware Log.txt](https://github.com/tier4/AWSIM/files/10557061/Autoware.Log.txt)

[image]
[image]
[image]

shmpwk, 2023-02-02:
@Fadimee 
Thank you for your investigation.
It looks your environment is organized as we expected.

> There are a new error on autoware output after reinstall autoware:
'/localization/initialize: status code 1 'The vehicle is not stopped.'

Autoware will not perform initial position estimation unless the vehicle has been stopped for at least 3 seconds.
This means that even if the vehicle has been stopped up to that point, initial position estimation will not be performed for 3 seconds after Autoware is activated. In some cases, the time to wait for route may be longer than before.

Could you set stop_check_enabled false 
https://github.com/autowarefoundation/autoware.universe/blob/ce93396d7b17dbb37666a7251d148fa1dc26aa4d/launch/tier4_localization_launch/launch/util/util.launch.xml#L19  to disable initial position estimation even if the vehicle is not stopped.

Fadimee, 2023-02-02:
@shmpwk 
'/localization/initialize: status code 1 'The vehicle is not stopped.'  Error solved with your solution but 2D Pose Estimate" still doesnt work. ı encountered wit a new error in output log;

`[component_container-64] [WARN] [1675318671.306431325] [control.trajectory_follower.lane_departure_checker_node]: failed to get transform from map to base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[component_container-64] [INFO] [1675318671.306494429] [control.trajectory_follower.lane_departure_checker_node]: waiting for self pose...
[component_container-62] [INFO] [1675318672.066189420] [planning.scenario_planning.parking.costmap_generator]: waiting for initial pose...
[rviz2-70] [INFO] [1675318675.175711744] [rviz2]: Setting estimate pose: Frame:map, Position(81990.9, 50469.4, 0), Orientation(0, 0, 0.0329564, 0.999457) = Angle: 0.0659246
[initial_pose_adaptor-67] [INFO] [1675318675.199432433] [default_ad_api.helpers.initial_pose_adaptor]: client call: /api/localization/initialize
[component_container_mt-65] [INFO] [1675318675.199795401] [default_ad_api.node.localization]: server call: /api/localization/initialize
[component_container_mt-65] [INFO] [1675318675.200257685] [default_ad_api.node.localization]: client call: /localization/initialize
[pose_initializer_node-32] [INFO] [1675318675.200525850] [localization.util.pose_initializer_node]: server call: /localization/initialize
[pose_initializer_node-32] [INFO] [1675318675.201731232] [localization.util.pose_initializer_node]: Deactivation succeeded
[pose_initializer_node-32] [INFO] [1675318675.201997108] [localization.util.pose_initializer_node]: Call NDT align server.
[ndt_scan_matcher-34] [WARN] [1675318675.202112729] [localization.pose_estimator.ndt_scan_matcher]: No InputSource
[pose_initializer_node-32] [INFO] [1675318675.202283754] [localization.util.pose_initializer_node]: NDT align server failed.
[pose_initializer_node-32] [INFO] [1675318675.202379953] [localization.util.pose_initializer_node]: server exit: /localization/initialize
[service_log_checker-3] [ERROR] [1675318675.202582418] [system.service_log_checker]: /localization/initialize: status code 4 'NDT align server failed.' (/localization/util/pose_initializer_node)
`

I am not sure you noticed beacuse I didnt mentioned about it in my previous posts,  I want to remind the map on autoware is not aligned correctly. You can see in picture below. I think that is the reason of issue. But I am not sure why it is not aligned and how I can fix it.

[image]

shmpwk, 2023-02-02:
@Fadimee 
>  ı encountered wit a new error in output log

It looks there is not the critical error.

As for the map, you first use AWSIM v1.0.1 but the map is different between that and v1.1.0 (or the latest version).
Could you confirm the map was updated?

Fadimee, 2023-02-02:
ıs there anything I can do for fixing map alignment? Because lane lines and point cloud are not aligned correctly.

[image]
[image]
[image]

shmpwk, 2023-02-02:
@Fadimee 

I've checked the latest map and found something wrong with it.
Now I'm quickly asking our team about it. 

[image]

shmpwk, 2023-02-02:
I continued to examine it but will reply tomorrow, Japan time.

shmpwk, 2023-02-02:
@isouf 
Thank you for your help. It looks your solution is [here](https://github.com/autowarefoundation/autoware/issues/3173#issuecomment-1399420623).

Fadimee, 2023-02-06:
@shmpwk and @isouf I checked both solutions but the problem still exist.
@shmpwk Are there any news about your examination?

shmpwk, 2023-02-10:
@Fadimee I could not found the cause of your problem yet.
Let me examine one by one. 
- Could you share the video from launching Autoware? 
- Could you share your Autoware launching command ?
- Do you use `awsim-stable` branch for launching Autoware?

shmpwk, 2023-02-17:
@isouf 
Looking at your log, lidar_centerpoint dies for example, which might have cuda related problem.

shmpwk, 2023-02-17:
You can also reffer to the other example of docker launching.
https://github.com/tier4/AWSIM/issues/91#issuecomment-1432854562

isouf, 2023-02-17:
Thank you @shmpwk. The `lidar_centerpoint` does not work because the `/sensing/lidar/top/pointcloud_raw` is not published, you can also see in the video above, Unity is not rendering the pointcloud. 

Looking at the `.config/unity3d/TIERIV/AWSIM/Player.log` I noticed the following error:

```bash
Unrecoverable error (code=500): cuda error: out of memory (code=2) @ /code/src/DeviceBuffer.hpp:127
```

The `nvidia-smi` output reveals that AWSIM consumes the entire GPU memory (~4GB):

```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.85.12    Driver Version: 525.85.12    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  On   | 00000000:01:00.0 Off |                  N/A |
| N/A   59C    P0    39W /  40W |   3893MiB /  4096MiB |    100%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2343      G   /usr/lib/xorg/Xorg                 20MiB |
|    0   N/A  N/A     49023    C+G   ..._v1.1.0/AWSIM_demo.x86_64     3862MiB |
+-----------------------------------------------------------------------------+
```

I believe this issue is also related to #87.

shmpwk, 2023-03-10:
@isouf 
Sorry for being late.
I tried Docker + AWSIM and looks they consumes 10GB with my Desktop :disappointed: 

```
$ nvidia-smi
Fri Mar 10 15:53:28 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 515.65.01    Driver Version: 515.65.01    CUDA Version: 11.7     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  On   | 00000000:01:00.0  On |                  N/A |
| 41%   71C    P0   224W / 250W |   8199MiB / 11264MiB |     98%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1208      G   /usr/lib/xorg/Xorg                532MiB |
|    0   N/A  N/A      1389      G   /usr/bin/gnome-shell               76MiB |
|    0   N/A  N/A      5370      G   ...146940414869142090,131072       25MiB |
|    0   N/A  N/A    903487      C   ...nt/lidar_centerpoint_node     1633MiB |
|    0   N/A  N/A    903534      C   ...ts/component_container_mt      877MiB |
|    0   N/A  N/A    903875      G   ...os/humble/lib/rviz2/rviz2       98MiB |
|    0   N/A  N/A    904445    C+G   ..._v1.1.0/AWSIM_demo.x86_64     4942MiB |
```

yangjiameng, 2023-03-10:
@shmpwk 
Hi，I have the same problem，2D Pose Estimate button doesnt work，I use AWSIM V1.1.0 and autoware galactic。

shmpwk, 2023-03-11:
@yangjiameng 
AWSIM v1.1.0 is for Ubuntu 22.04, Humble.
When you use Ubuntu 22.04, Galactic, please try AWSIM v1.0.2.

yangjiameng, 2023-03-11:
@shmpwk 
Thank you！And The topic don‘t  appear by “ros2 topic list” when I use AWSIM v1.0.1。
And I have written “export ROS_LOCALHOST_ONLY=1
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

if [ ! -e /tmp/cycloneDDS_configured ]; then
    sudo sysctl -w net.core.rmem_max=2147483647
    sudo ip link set lo multicast on
    touch /tmp/cycloneDDS_configured
fi”
in ～/.bashrc file.
But the AWSIM is breakdown

shmpwk, 2023-03-11:
> the AWSIM is breakdown

What do you mean by breakdown? The simulation is crashed ? 
If your PC don't have sufficient memory, it sometimes happens.

FYI, I created AWSIM + docker instruction [here](https://github.com/tier4/autoware-online-document)

Mufan187569, 2023-03-11:
@shmpwk 
Hello,I have the same problem as yangjiameng .
AWSIM broken down,which means that it flashed out.

Mufan187569, 2023-03-11:
@shmpwk 
And I just downloaded AWSIM v1.0.2, but I do not know how to use it. Do you have a tutorial for using it?

shmpwk, 2023-03-11:
@Mufan187569 

For Ubuntu 20.04, Galactic, you can refer to the v1.0.2 tutorial. 
https://github.com/tier4/AWSIM/blob/v1.0.2/docs/GettingStarted/QuickStartDemo/index.md

Mufan187569, 2023-03-11:
@shmpwk 
 Thank you very much for sharing the tutorial with me. The AWSIM_demo.x86._64 mentioned in the tutorial is not found in v1.0.2, so how should I start AWSIM

shmpwk, 2023-03-11:
@Mufan187569 
In the tutorial, there is a description.
The you can download AWSIM_V1.0.1 binary in AWSIM ver 1.0.2

[image]

yangjiameng, 2023-03-12:
@shmpwk 
OK！ Thank you! I have successfully run autoware galactic and AWSIM v 1.0.1 in ubuntu 20.04.
But the planning no route。

Mufan187569, 2023-03-13:
@shmpwk 
Thank you, the vehicle can be initially positioned. But now there is a problem that when setting the goal position, the vehicle status shows that it is planning, not driving

Mufan187569, 2023-03-14:
@shmpwk 
Thank you. I have successfully run Autoware with AWSIM. I want to ask about the topic of AWSIM release,What's the difference between /sensing/lidar/top/pointcloud_raw and /sensing/lidar/top/pointcloud_ raw_ ex?
And now we are deploying autoware to real vehicles. What should we do to replace AWSIM radar？

shmpwk, 2023-05-01:
@yangjiameng @Mufan187569 
Sorry for too late reply. How' s it going?

> What's the difference between /sensing/lidar/top/pointcloud_raw and /sensing/lidar/top/pointcloud_ raw_ ex?

The latter one is extracted one. [source](https://github.com/autowarefoundation/velodyne_vls/blob/tier4/universe/velodyne_pointcloud/src/conversions/convert.cc).

> And now we are deploying autoware to real vehicles. What should we do to replace AWSIM radar？

If you integrate real vehicle, you can ignore the radar. You can switch not to use radar in Autoware launch file.

zhuyingxiang95, 2023-05-15:
i got the almost the same issue. could someone give me some help.  log file attached here.
[launch.log](https://github.com/tier4/AWSIM/files/11476906/launch.log)

shmpwk, 2023-05-15:
@zhuyingxiang95 Seems to be solved https://github.com/tier4/AWSIM/issues/129

manikandanhu2016, 2024-08-14:
I also have same issue can you help to solve this problem 
environment details:
ROS2 HUMBLE
AWSIM 1.2.3.
Ubuntu 22
Autoware. Universe
[image]
[image]
[image]
[image]

thknx, 2025-02-06:
Hi guys, I have same issue too