autowarefoundation/AWSIM #77: Please confirm the ubuntu version for quick demo (pose estimation problem)

Garithe, 2023-01-24:
First, while the galactic version is required, it can not be installed on the ubuntu22. But the quick demo page specifies
Ubutnu 22.04
Which OS exactly works? 
Plus, I tried ubuntu 20 + autoware(galactic) + awsim, but autoware doesn't run self pose estimation (quick demo).  
I followed:
https://github.com/tier4/AWSIM/blob/v1.0.2/docs/GettingStarted/QuickStartDemo/index.md

shmpwk, 2023-01-25:
@Garithe 
Thank you for pointing that out.
I had described the docs incorrectly.
I fixed the docs [here](https://github.com/tier4/AWSIM/pull/78).

So the correct combination is
v1.1.0 ~ latest main branch:  Ubuntu22 + humble
~ v1.0.2: Ubuntu20 + galactic

shmpwk, 2023-01-25:
> Plus, I tried ubuntu 20 + autoware(galactic) + awsim, but autoware doesn't run self pose estimation (quick demo, 5-2). I followed: https://github.com/tier4/AWSIM/blob/v1.0.2/docs/GettingStarted/QuickStartDemo/index.md

Your environment and procedures seem to correct.
Do you have some error when `./setup-dev-env.sh` or `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-w"` ?

Garithe, 2023-01-25:
@shmpwk 
I think I waited for a while..
How to know if it launches well?
I think I waited until the image is similar to that in instruction.

shmpwk, 2023-01-25:
> How to know if it launches well?

I recommend to grep error
```
ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit map_path:=<your mapfile location> | grep error
```
```
ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit map_path:=<your mapfile location> | grep die
```
If you get some outputs when executing the above command, could you share them?

shmpwk, 2023-01-25:
Could you share your picture of RViz to judge the whether it is ok?

Garithe, 2023-01-25:
@shmpwk Sorry I have reinstalled OS for some reason. 
But I remember that the AutowareStatePanel show INT for all items. 
And I tried to add start pose and goal by somewhat 2d pose goal or something, the car is now shown on map as expected.

shmpwk, 2023-01-26:
>  the car is now shown on map as expected
You mean your problem has been solved? 
If so, could you close this issue?

shmpwk, 2023-01-26:
> The previous reply was wrong. I meant the car was NOT shown on map.

Ok, then could you share https://github.com/tier4/AWSIM/issues/77#issuecomment-1403860336 or https://github.com/tier4/AWSIM/issues/77#issuecomment-1403863634 after your reinstallation is completed.
With the current information, I have no idea what could be causing this.

shmpwk, 2023-02-10:
@Garithe How is it going?

shmpwk, 2023-02-16:
@Garithe Please open this issue again if you still have this trouble.

closed 2023-02-16 by shmpwk