autowarefoundation/AWSIM #80: Start the Vehicle

joseph16388, 2023-01-29:
Now I could start unity and simulator smoothly. When I selected 2d global pose according to the tutorial, routing trajectories did occur, but it could not automatically start. What is the reason for this?
thank U!
the screen shot is below:
[image]

shmpwk, 2023-01-29:
@joseph16388 
You have to command to start the vehicle by ros2 service as the tutorial says.

joseph16388, 2023-01-29:
> @joseph16388 You have to command to start the vehicle by ros2 service as the tutorial says.

Yes,after I engae,vehicle could run,thanks!

closed 2023-01-29 by joseph16388