autowarefoundation/AWSIM #86: About awsim simulator

joseph16388, 2023-02-08:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other [issues](https://github.com/tier4/AWSIM/issues?q=is%3Aissue) and [discussions Q&A category](https://github.com/orgs/autowarefoundation/discussions/categories/q-a). I found no duplicate issues.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I downloaded awsim and entered the autowaresimulation scene, and found that the car would fall, but I didn't see any Japanese street view during the co-simulation with autoware, please tell me what is the situation, thank you!

### Expected behavior

As in the co-simulation, the vehicle waits for departure

### Actual behavior

The vehicle is in free fall in AWSIM simulator

### Steps to reproduce

1、open awsim
2、enter autowaresimulation scene
3、The vehicle is in free fall

### Versions

ROS2
Autoware Universal
Awsim

shmpwk, 2023-02-10:
@joseph16388 
Does it mean that the car will fall off if it runs through the end of the road?
If so, we don't support the behavior beyond the defined road.
If my response does not make sense, could you share the video of free fall?

shmpwk, 2023-02-10:
I update the title to describe the issue clearly

joseph16388, 2023-02-11:
> @joseph16388 Does it mean that the car will fall off if it runs through the end of the road? If so, we don't support the behavior beyond the defined road. If my response does not make sense, could you share the video of free fall?
Hi！@shmpwk
[image]

shmpwk, 2023-02-12:
It looks our undefined scene on the map which we does not support.

joseph16388, 2023-02-13:
> It looks our undefined scene on the map which we does not support.

Hello!
I want to open the default scenario(japanese street map) for awsim and autoware co-emulation. What is the name of this scenario?
Thank you!

shmpwk, 2023-02-13:
@joseph16388 
> default scenario(japanese street map) 

What do you mean by scenario? https://github.com/tier4/scenario_simulator_v2 ?
I think you have already completed the tutorial: https://github.com/tier4/AWSIM/issues/80 but you want to do more?

joseph16388, 2023-02-14:
> 

scenario_simulator_v2 is the simulator of the scenario below?
[image]

shmpwk, 2023-02-14:
No it is like this https://tier4.github.io/scenario_simulator_v2-docs/user_guide/QuickStart/ .
(It is not related to AWSIM but we often call scenario_simulator_v2 as a simulator of the scenario) 

What do you mean by scenario ?

joseph16388, 2023-02-14:
> No it is like this https://tier4.github.io/scenario_simulator_v2-docs/user_guide/QuickStart/ . (It is not related to AWSIM but we often call scenario_simulator_v2 as a simulator of the scenario)
> 
> What do you mean by scenario ? Do you ask us whether we can try scenario simulation on AWSIM?

1、I just want to know awsim simulator executable default scenario, corresponding to the https://github.com/tier4/AWSIM which scene? I want to modify the scene and script according to https://github.com/tier4/AWSIM

2、scenario_simulator_v2 is another autoware simulation simulator？And what is the difference between https://github.com/tier4/AWSIM simulato and scenario_simulator_v2 , as the direct support mentioned in this repo autoware version?
thank you！

shmpwk, 2023-02-14:
@joseph16388
1.  
Is the scenario you mean is that pedestrians and vehicles run randomly?
If so, you can reffer here https://github.com/tier4/AWSIM/tree/main/docs/Components/Traffic.

2. AWSIM simulation is for end to end simulation. On the other hands, scenario_simulation_v2 is for planning simulation only.

joseph16388, 2023-02-14:
> @joseph16388
>
> 1. Is the scenario you mean is that pedestrians and vehicles run randomly?
>    If so, you can reffer here https://github.com/tier4/AWSIM/tree/main/docs/Components/Traffic.
> 2. AWSIM [...]

Hello!
I mean in the autowaresimulation scenario I mentioned above, why would the road vehicles and pedestrians disappear? If it's not this scene, which scene is it and what's the name of it?

shmpwk, 2023-02-16:
I finally understand that you want to simulate in **unity environment**. (I mistakenly think that you want to simulate in RViz.)

If so did you try "Run the demo in Editor" https://tier4.github.io/AWSIM/GettingStarted/SetupUnityProject/ ?

joseph16388, 2023-02-16:
> @joseph16388 I finally understand that you want to simulate in **unity environment**. (I mistakenly think that you want to simulate in RViz.) If so did you try "Run the demo in Editor" https://tier4.github.io/AWSIM/GettingStarted/SetupUnityProject/ ?

yeah！I will try,thank you!

closed 2023-02-17 by joseph16388