autowarefoundation/AWSIM #87: Problem with running AWSIM binary v1.1.0 on Ubuntu22

soblin, 2023-02-08:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other [issues](https://github.com/tier4/AWSIM/issues?q=is%3Aissue) and [discussions Q&A category](https://github.com/orgs/autowarefoundation/discussions/categories/q-a). I found no duplicate issues.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I downloaded `AWSIM_v1.1.0.zip`, extracted, `chmod +x` , and then executed the binary from the terminal or by double clicking the icon. But it crashes:

```
Set current directory to /home/<user>/workspace/AWSIM_v1.1.0
Found path: /home/<user>/workspace/AWSIM_v1.1.0/AWSIM_demo.x86_6
Segmentation fault (core dumped)
```

According to [Trouble shooting](https://tier4.github.io/AWSIM/DeveloperGuide/TroubleShooting/) guide, this will be a case that I need to "Check if yourNvidia drivers or Vulkan API are installed correctly". Here is the output of `nvidia-smi`

```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 515.76       Driver Version: 515.76       CUDA Version: 11.7     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
| N/A   41C    P0    11W /  N/A |    648MiB /  4096MiB |     27%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2874      G   /usr/lib/xorg/Xorg                328MiB |
|    0   N/A  N/A      3925      G   /usr/bin/gnome-shell               64MiB |
|    0   N/A  N/A     98432      G   ...765856052798007594,131072      164MiB |
|    0   N/A  N/A    393964      G   ...veSuggestionsOnlyOnDemand       70MiB |
+-----------------------------------------------------------------------------+
```

`libvulkan1` is already installed, so I think Graphics is OK, but still the binary  crashes. I'm not sure if this information is useful, but here is the output of gdb:

```
(gdb) r ./AWSIM_demo.x86_64
...
...
[New Thread 0x7fff08076640 (LWP 397068)]
Thread 3 "Finalizer" received signal SIGPWR, Power fail/restart.
[Switching to Thread 0x7ffff0845640 (LWP 397002)]
(gdb) bt
__futex_abstimed_wait_common64 (private=<optimized out>, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x7ffff1168880) at ./nptl/futex-internal.c:57
57      ./nptl/futex-internal.c: No such file or directory.
(gdb) bt
#0  __futex_abstimed_wait_common64 (private=<optimized out>, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x7ffff1168880) at ./nptl/futex-internal.c:57
#1  __futex_abstimed_wait_common (cancel=true, private=<optimized out>, abstime=0x0, clockid=0, expected=0, futex_word=0x7ffff1168880) at ./nptl/futex-internal.c:87
#2  __GI___futex_abstimed_wait_cancelable64 (futex_word=futex_word@entry=0x7ffff1168880, expected=expected@entry=0, clockid=clockid@entry=0, abstime=abstime@entry=0x0,
    private=<optimized out>) at ./nptl/futex-internal.c:139
#3  0x00007ffff5c06c5f in do_futex_wait (sem=sem@entry=0x7ffff1168880, abstime=0x0, clockid=0) at ./nptl/sem_waitcommon.c:111
#4  0x00007ffff5c06cf8 in __new_sem_wait_slow64 (sem=0x7ffff1168880, abstime=0x0, clockid=0) at ./nptl/sem_waitcommon.c:183
#5  0x00007ffff0e144c2 in ?? () from /home/msobue/workspace/AWSIM_v1.1.0/AWSIM_demo_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so
#6  0x00007ffff0e146f1 in ?? () from /home/msobue/workspace/AWSIM_v1.1.0/AWSIM_demo_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so
#7  0x00007ffff0e14846 in ?? () from /home/msobue/workspace/AWSIM_v1.1.0/AWSIM_demo_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so
#8  0x00007ffff0de5cea in ?? () from /home/msobue/workspace/AWSIM_v1.1.0/AWSIM_demo_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so
#9  0x00007ffff0e47d1b in GC_inner_start_routine () from /home/msobue/workspace/AWSIM_v1.1.0/AWSIM_demo_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so
#10 0x00007ffff0e3c851 in GC_call_with_stack_base () from /home/msobue/workspace/AWSIM_v1.1.0/AWSIM_demo_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so
#11 0x00007ffff5bfeb43 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#12 0x00007ffff5c90a00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

### Expected behavior

The simulator starts.

### Actual behavior

It crashes on startup.

### Steps to reproduce

I'm not sure if this problem reproduces on other machines.

### Versions

- OS: Ubuntu22
- NVIDIA: displayed above

shmpwk, 2023-02-10:
@mackierx111 Could you assign someone for help?

kasper-helm, 2023-02-10:
I was having the same issue but was able to overcome it by following the instructions here https://github.com/tier4/AWSIM/issues/81

soblin, 2023-02-11:
@kasper-helm Wow that's good hear that ! I'll try it out

soblin, 2023-02-20:
I could run AWSIM on docker following this https://github.com/tier4/AWSIM/issues/81 (but still not on local machine).

Cheny5863, 2023-04-06:
> I could run AWSIM on docker following this #81 (but still not on local machine).

Hi bro. You shouldn't close this problem.I met it as same as you exactly . That is a bug.

wep21, 2023-04-26:
@Cheny5863 I found out that we need libssl1.0. Could you install the libraries as below?
```
wget http://security.ubuntu.com/ubuntu/pool/main/o/openssl1.0/libssl1.0.0_1.0.2n-1ubuntu5.10_amd64.deb
sudo dpkg -i libssl1.0.0_1.0.2n-1ubuntu5.10_amd64.deb
wget http://security.ubuntu.com/ubuntu/pool/main/o/openssl1.0/libssl1.0-dev_1.0.2n-1ubuntu5.12_amd64.deb
sudo dpkg -i libssl1.0-dev_1.0.2n-1ubuntu5.12_amd64.deb
```

yuasabe, 2023-04-27:
I seem to be facing the same problem. My machine specs are as follows:
- Ubuntu 22.04.1 LTS
- Linux 5.19.17-051917-generic x86_64
- NVIDIA driver: 525.105.17
- GPU: GeForce RTX 3080 Ti Mobile
- ROS 2 Humble

wep21, 2023-04-27:
In my environment, after I once installed openssl1.0.0, the issue seems to disappeared.
```
wget http://security.ubuntu.com/ubuntu/pool/main/o/openssl1.0/libssl1.0.0_1.0.2n-1ubuntu5.12_amd64.deb
sudo dpkg -i libssl1.0.0_1.0.2n-1ubuntu5.12_amd64.deb
```

Zoe-Wan, 2023-05-13:
I have met the very same problem when I am using WSL2 in Windows10. In my case, it is because WSL2 has not yet supported Vulkan API, and the related imformation (about the WSL2 and Vulkan) can be found in https://github.com/microsoft/WSL/issues/7790 .

lee-devin, 2023-06-05:
It's like vulkan-sdk encouter problem, but I can't install it, any one solved the issue now?

szepilot, 2023-06-14:
Same problem here: OS: Ubuntu 22.04, ROS 2 Humble, Nvidia GTX 1050 Ti

``` r
vainfo

libva info: VA-API version 1.14.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```
``` r
nvidia-smi 

Wed Jun 14 13:46:30 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.116.04   Driver Version: 525.116.04   CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:65:00.0  On |                  N/A |
| 41%   37C    P0    N/A /  75W |    537MiB /  4096MiB |      2%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1912      G   /usr/lib/xorg/Xorg                188MiB |
|    0   N/A  N/A      2078      G   /usr/bin/gnome-shell              115MiB |
|    0   N/A  N/A      2789      G   ...5/usr/lib/firefox/firefox      202MiB |
+-----------------------------------------------------------------------------+

```

tksino7891, 2023-08-06:
I had the same issue.
I was able to run after doing following.
```
export ROS_LOCALHOST_ONLY=1
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
sudo sysctl -w net.core.rmem_max=2147483647
sudo ip link set lo multicast on
touch /tmp/cycloneDDS_configured
sudo add-apt-repository ppa:graphics-drivers/ppa
sudo apt update
sudo ubuntu-drivers autoinstall
```
Not sure what is needed.
However, after last command `ubuntu-drivers autoinstall` AWSIM was able to run.

tksino7891, 2023-08-13:
I have tested many time for building autoware environment.
There seem to be some problem for the official set up.
The environment in `autoware/amd64.env` is as follows.
```
rosdistro=humble
rmw_implementation=rmw_cyclonedds_cpp
base_image=ubuntu:22.04
cuda_base_image=ubuntu:22.04
prebuilt_base_image=ubuntu:22.04
cuda_version=11.6
cudnn_version=8.4.1.50-1+cuda11.6
tensorrt_version=8.4.2-1+cuda11.6
```
cuda 11.7 is supported from ubuntu 22.04. 11.6 is not supported.
cudnn8.4 not supprted for ubuntu 22.04
tensorrt8.4 not supported for ubuntu 22.04
There seems to be problem for the supported driver.

Autostone-c, 2023-09-07:
i have the same issue
but Segmentation fault (core dumped) not appear. only like this 
[image]

Autostone-c, 2023-09-11:
hi , i try it few times . i found my fault because of the autoware(awsim-stable branch) setup-dev-env.sh。
i follow the https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/  tutorial can work。but after i run setup-dev-env.sh . the AWSIM demo can not work. anybody has ideas?

Sam827-r, 2024-01-23:
@ szepilot  did you resolved your issue i have same Machine specs

szepilot, 2024-01-23:
> @szepilot did you resolved your issue i have same Machine specs

No, but works on my other PC with a newer GPU

Sam827-r, 2024-01-24:
@soblin @soblin 

> I could run AWSIM on docker following this #81 (but still not on local machine).

can you please specify your machine specifications.

Sam827-r, 2024-01-31:
@

> I have tested many time for building autoware environment. There seem to be some problem for the official set up. The environment in `autoware/amd64.env` is as follows.
>
> ```
> rosdistro=humble [...]
did you find any solution on it .

szepilot, 2024-03-18:
> @soblin @szepilot
> 
> > I could run AWSIM on docker following this #81 (but still not on local machine).
> 
> can you please specify your newer machine specifications along with RAM,CPU,GPU config .

NVIDIA RTX A4000, Intel(R) Xeon(R) W-2125 CPU @ 4.00GHz, 64 GB RAM

ShoukatM, 2024-04-30:
its working on RTX 4070.

closed 2025-08-07 by mackierx111