autowarefoundation/AWSIM #92: null pointer error for awsim

joseph16388, 2023-02-17:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other [issues](https://github.com/tier4/AWSIM/issues?q=is%3Aissue) and [discussions Q&A category](https://github.com/orgs/autowarefoundation/discussions/categories/q-a). I found no duplicate issues.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The AutowareSimulation scene of unity when press the play button, a null pointer error is reported. The screenshot is as follows:
[image]

### Expected behavior

Null pointer error is not reported

### Actual behavior

null pointer error is reported

### Steps to reproduce

enter awsim autowaresimulation scene and press play

mackierx111, 2023-02-28:
@joseph16388 
It appears that ROS2 in AWSIM is not working properly.

Is there any reference to troubleshooting here? If there is, I'd like you to try it first.
https://tier4.github.io/AWSIM/DeveloperGuide/TroubleShooting/

joseph16388, 2023-03-01:
> @joseph16388 It appears that ROS2 in AWSIM is not working properly.
> 
> Is there any reference to troubleshooting here? If there is, I'd like you to try it first. https://tier4.github.io/AWSIM/DeveloperGuide/TroubleShooting/

Thanks for your reply! But the awsim binary version doesn't have this problem with ROS2

mackierx111, 2023-03-01:
Which OS, windows or ubuntu?
Can you check at the beginning of the log to see if  `C:\ci\ws\src\ros2\rcl\rcl\src\rcl\node.c:262` errors are shown?

joseph16388, 2023-03-02:
> Which OS, windows or ubuntu? Can you check at the beginning of the log to see if `C:\ci\ws\src\ros2\rcl\rcl\src\rcl\node.c:262` errors are shown?

hello, I am using ubuntu20

amadeuszsz, 2023-03-03:
> > Which OS, windows or ubuntu? Can you check at the beginning of the log to see if `C:\ci\ws\src\ros2\rcl\rcl\src\rcl\node.c:262` errors are shown?
> 
> hello, I am using ubuntu20

Same error here, but I know the reason in my case, maybe it will be helpful for you. As you work with Autoware in docker container (**if not, I have no idea what cause your issues**), all the necessary dependencies are not visible on host system, unless you run Unity inside docker, but I don't think so. Errors will disappear if you install Autoware on host machine. Autoware supports humble distro now so you need upgrade your OS or switch to galactic branch.

mackierx111, 2023-03-06:
AWSIM v1.1.0 and above only supports ros2 humble. please try it on Ubuntu 22.04

closed 2023-03-06 by joseph16388

yaukaizhi, 2023-07-10:
[image]

I have encountered the same issue, I'm using Source build on Ubuntu22 with Humble.