autowarefoundation/AWSIM #93: Unity simulation of AWSIM not working

MarioAlmelaUMH, 2023-02-20:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other [issues](https://github.com/tier4/AWSIM/issues?q=is%3Aissue) and [discussions Q&A category](https://github.com/orgs/autowarefoundation/discussions/categories/q-a). I found no duplicate issues.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When I tried the quick demo version, it work fine without any noticeable bugs, but when I tried to setup the project in Unity, following the steps shown in https://tier4.github.io/AWSIM/GettingStarted/SetupUnityProject/, the ego vehicle does not automatically set its initial position. I've tried following the troubleshooting and setting up the initial position manually, but the 2D Pose Estimate button does not spawn the ego vehicle anywhere.

### Expected behavior

The ego vehicle sets its initial position and everything works like in the quick demo version.

### Actual behavior

The ego vehicle does not appear in Autoware RViz and is impossible to set up manually.

### Steps to reproduce

I don't know how to reproduce the issue or if more people have experimented it.

### Versions

- OS: Ubuntu 22.04
- ROS: ROS 2 Humble
- AWSIM: 1.1.0

### Possible causes

I don't know exactly the causes, but the ego vehicle doesn't set up its initial position after launching Autoware and the Unity simulation. May be the Unity simulation can not communicate the vehicle's initial position properly to Autoware.

### Additional context

[image]

[image]

LiangHuangBC, 2023-02-22:
same issue on released binary, error is under system->vehicle, no transform from [base_link] to [map]

mackierx111, 2023-02-28:
@LiangHuangBC 
Could you please try to source ros2 before starting UnityEditor?

ga58lar, 2023-03-02:
I am using the same OS, ROS2 and AWSIM version and did not experience this issue.   
My only difference from the instructions is that I use Fast-DDS instead of cyclone-DDS.  
  
Are the ROS2 params (e.g. domain_id, localhost) set correctly? Did you check if you see any ROS2 topics and if you are receiving ROS2 messages (e.g. ros2 topic list)?

closed 2023-03-03 by MarioAlmelaUMH

zhuyingxiang95, 2023-05-11:
it may caused for ROS 2 work improperly