autowarefoundation/AWSIM #96: LiDAR sensor "apply gaussian noise" not working

ga58lar, 2023-02-28:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other [issues](https://github.com/tier4/AWSIM/issues?q=is%3Aissue) and [discussions Q&A category](https://github.com/orgs/autowarefoundation/discussions/categories/q-a). I found no duplicate issues.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The parameter _Apply Gaussian Noise_ in the LiDAR sensor configuration in Unity Editor does not seem to work as intended, as no noise will be added to the LiDAR points with the enabling of the said parameter.
Also the toggling of the box sets the variable in the LidarSensor.cs file to true, but the parameter seems to not be referenced by any other file.
[image]

### Expected behavior

Enabling of gaussian noise should add noise to the LiDAR points.

### Actual behavior

LiDAR points stay perfectly static, even with enabled gaussian noise and independent of the chosen noise params.

### Steps to reproduce

1. Follow instructions of "Setup Unity Project"
2. Toggle the Gaussian Noise box in Unity Editor
3. Play scenario in Unity Editor

### Versions

-OS: Ubuntu 20.04
-AWSIM: v1.0.2
&
-OS: Ubuntu 22.04
-AWSIM: v1.1.0

prybicki, 2023-03-06:
@ga58lar Thank you for the report; your observation is correct. This will be fixed in the upcoming RobotecGPULidar upgrade within a few weeks (v0.13). There is already a draft PR that addresses this issue (#88). Please let me know if this feature is important to you, and we may consider expediting it.

The parameter (which should have been removed) is a leftover from the previous implementation of the Gaussian Noise feature. The upcoming one allows adding different kinds of noise independently (angular, distance) and will come with some bug fixes.

ga58lar, 2023-03-07:
@prybicki  Thank you for the reply and the information.  
I would of course appreciate a expedited implementation, if that is possible from your side as I am currently looking into the applicability of the simulation for our use case, where the LiDAR model plays a crucial role.  
I will have a look at the PR and see if I can already use the feature-branch as an temporal solution. Thanks.

prybicki, 2023-03-07:
@ga58lar Ok, we will expedite these efforts. The branch should work now (let me know if it doesn't). In a few days, we will push the most recent RGL binaries and final changes to improve the user interface. I will also let you know when this feature is merged into the main AWSIM branch, and then we'll close the issue.

prybicki, 2023-03-09:
@ga58lar We have pushed some UI changes and Linux binaries to [feature/rgl-gaussian-noise-reimplementation](https://github.com/tier4/AWSIM/tree/feature/rgl-gaussian-noise-reimplementation) branch. We need to run a few chores (e.g. build Windows binaries) before merging into `main`, but there shouldn't be any substantial changes.

prybicki, 2023-04-13:
Solved in #88

closed 2023-04-13 by prybicki