autowarefoundation/AWSIM #98: Parent objects get shifted when adding camera sensor from prefab

PascalBlumIntech, 2023-03-07:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other [issues](https://github.com/tier4/AWSIM/issues?q=is%3Aissue) and [discussions Q&A category](https://github.com/orgs/autowarefoundation/discussions/categories/q-a). I found no duplicate issues.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Adding a camera sensor from prefab causes an arbitrary shifting of objects that are at higher hierarchy levels than the camera sensor. The issue is reproducible as it happened on two different computers. There are no custom scripts used, the issue occurs when following the instruction on https://tier4.github.io/AWSIM/GettingStarted/SetupUnityProject/.

### Expected behavior

Adding a camera sensor should not have any effects on the positioning of superordinated objects.

### Actual behavior

After completing the reproduction steps:

Take a look to superordinated objects like sensor_kit_base_link or URDF and remember their position shown in scene view. Now, deactivate CameraSensor and check the position of those superordinated objects in the scene view. We have observed that the object moves its parent objects. Take a look to those screenshots, please:

[image]
[image]

However, it is noticeable that the shifts are not traceable with the inspector. This table shows how the positions of the relevant hierarchical ordered objects are conditioned (rounded to two decimal places):

[image]

There is no difference, so the shift cannot be traced for unknown reasons.
Adding a second CameraSensor [named “CameraSensor (1)”, local position vector3(0, 1, -10)] leads to the following shifts:

[image]
[image]
[image]
[image]

### Steps to reproduce

We followed the instructions at https://tier4.github.io/AWSIM/GettingStarted/SetupUnityProject/. Accordingly, after installing the correct Unity installation 2021.1.7f1, we cloned the git@github.com:tier4/AWSIM.git repository. We then opened it with Unity. As mentioned in the setup documentation, we downloaded Japan_Tokyo_Nishishinjuku.unitypackage to add it within Unity.
In the scene AutowareSimulation (Project Explorer: Assets/AWSIM/Scenes/Main/AutowareSimulation.unity) there is a Lexus RX450h (Hierarchy: AutowareSimulation/EgoVehicle/Lexus RX450h 2015 Sample Sensor). Subordinated to the Lexus object is the partial hierarchy URDF/base_link/sensor_kit_base_link. We added to sensor_kit_base_link an empty game object called sensor_base_project as a child of sensor_kit_base_link. 
The object sensor_base_project should be the base of all the other sensors we plan to add. That base should be positioned at the center of the rear axis. The local position we used for sensor_base_project is Vector3(0, -1.67905247, -0.850758076). There is no rotation or scale applied.
In the scene view, sensor_base_project is positioned right in the center of the rear axis as expected.  Subsequently, we decided to bundle all camera sensors under a common parent object. We called it cameras_project and made it a child of sensor_base_project. Then, we added one camera sensor as a child of camera_project by drag&drop, using the camera sensor prefab (Project Explorer: Assets/AWSIM/Prefabs/Sensors/CameraSensor.prefab). 
This screenshot shows the hierarchy at this point:

[image]

CameraSensor had a position offset vector3(0, 1, -10) as the default setting that we removed to vector3(0, 0, 0). Now, the problem can be experienced as mentioned in section "Actual behaviour". 

### Versions

The issue occurred first on an academical GPU Server (Ubuntu 20.04). We set up a second computer to eliminate the complexity of the environment. These are the specifications:
•	Linux Ubuntu 20.04.1
•	Intel i7-8700k 12x3,7 GHz
•	32GB RAM DDR4
•	Geforce GTX 1070ti 8GB
•	Unity Version 2021.1.7f1

No Autoware or ROS involved.

mackierx111, 2023-03-13:
Thanks for the report. I will check.

PascalBlumIntech, 2023-05-03:
Hi! @mackierx111 Are you still on it? Thank you!

mackierx111, 2023-05-09:
@PascalBlumIntech 
I think it might be the Unity pivot and center specs.

> For position
Use the dropdown menu to switch between Pivot and Center.
Pivot positions the Gizmo at the actual pivot point of the GameObject, as defined by the Transform component.
Center positions the Gizmo at a center position based on the selected GameObjects.

https://docs.unity3d.com/Manual/PositioningGameObjects.html

mackierx111, 2023-05-09:
Also, since AWSIM uses RTX ray tracing in lidar simulation, an RTX graphics card is preferred.

closed 2023-06-03 by mackierx111