autowarefoundation/AWSIM #99: The dynamic vehicle visuals glitch for some reason.

jediofgever, 2023-03-07:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other [issues](https://github.com/tier4/AWSIM/issues?q=is%3Aissue) and [discussions Q&A category](https://github.com/orgs/autowarefoundation/discussions/categories/q-a). I found no duplicate issues.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The dynamic vehicle visuals glitch for some reason. 
 

### Expected behavior

The visual glitch should not happen. 

### Actual behavior

The dynamic vehicle visuals glitch for some reason. 

### Steps to reproduce

With given versions of libraries, glitching happens. 

### Versions

- Ubuntu 22.04
- Unity Editor, 2021.3.20f1
- Humble, ROS2

### Possible causes

It could be that I am using different unity editor but apart from that everything functions as expected. 

https://user-images.githubusercontent.com/18460336/223550971-f7771750-f04c-4c4f-8b2e-d2e80d6bd210.mp4

mackierx111, 2023-03-13:
Perhaps using a different Unity version may be the cause. Could you use Unity 2021.1.7f1?

jediofgever, 2023-03-24:
The results are same, 
I also get some errors complaining that ros libraries cannnot be found, 

```
NullReferenceException: Object reference not set to an instance of an object
AWSIM.ImuRos2Publisher.Publish (AWSIM.ImuSensor+OutputData outputData) (at Assets/AWSIM/Scripts/Sensors/Imu/ImuRos2Publisher.cs:86)
AWSIM.ImuSensor.FixedUpdate () (at Assets/AWSIM/Scripts/Sensors/Imu/ImuSensor.cs:100)
```

mackierx111, 2023-05-09:
@jediofgever 
Sorry for the late reply.
Can you tell us what PC specs you are using?

mackierx111, 2023-06-19:
I'll close it once.

closed 2023-06-19 by mackierx111