autowarefoundation/AWSIM #126: Unable to drive in reverse

hir-yk, 2023-05-10:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other [issues](https://github.com/tier4/AWSIM/issues?q=is%3Aissue) and [discussions Q&A category](https://github.com/orgs/autowarefoundation/discussions/categories/q-a). I found no duplicate issues.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When planning for free space in the parking area and initiating reverse driving, my vehicle stops moving. 

### Expected behavior

The vehicle should drive in reverse.

### Actual behavior

The vehicle does not move.

https://github.com/tier4/AWSIM/assets/84175137/eba5cd46-e07d-434b-b191-6a1e1f4acecc

### Steps to reproduce

1. Set the parking space as the Goal. 
2. Start autonomous driving. 

### Versions

- OS: Ubuntu22.04
- ROS: Humble
- Autoware: Universe
- AWSIM: 1.1.0

mackierx111, 2023-05-23:
@hir-yk 
Thanks for the report. I have corrected the behavior with a pull request [here](https://github.com/tier4/AWSIM/pull/131). Merged into main branch.

hir-yk, 2023-05-29:
Great!  My vehicle can now move in reverse. Thank you.

closed 2023-05-29 by hir-yk