autowarefoundation/AWSIM #129: the initial pose could not be set

zhuyingxiang95, 2023-05-15:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other [issues](https://github.com/tier4/AWSIM/issues?q=is%3Aissue) and [discussions Q&A category](https://github.com/orgs/autowarefoundation/discussions/categories/q-a). I found no duplicate issues.
- [X] I'm convinced that this is not my fault but a bug.

### Description

the initial pose could not neither match by itself, nor set up manually in Rviz. 
[launch.log](https://github.com/tier4/AWSIM/files/11477120/launch.log)
Log file attached

### Expected behavior

the initial pose in Autoware and Awsim will automatically match 

### Actual behavior

the initial pose could  not be set in Rviz. 

### Steps to reproduce

run Awsim binary  and then run autoware

### Versions

Ubuntu22.04 
ros2 Humble

shmpwk, 2023-05-15:
Error log 

```
1684145744.0423956 [component_container-61] [ERROR] [1684145742.855303019] [planning.scenario_planning.parking.parking_container]: Component constructor threw an exception: could not create subscription: rcl node's context is invalid, at ./src/rcl/node.c:428
1684145744.0742576 [INFO] [component_container_mt-60]: process has finished cleanly [pid 7611]
1684145744.1532190 [INFO] [component_container-39]: process has finished cleanly [pid 7409]
1684145744.1789815 [INFO] [component_container-64]: process has finished cleanly [pid 7711]
1684145744.1902704 [INFO] [component_container_mt-65]: process has finished cleanly [pid 7781]
1684145744.2605820 [INFO] [component_container_mt-58]: process has finished cleanly [pid 7595]
1684145749.0325949 [ERROR] [rviz2-70]: process[rviz2-70] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
1684145749.0330775 [INFO] [rviz2-70]: sending signal 'SIGTERM' to process[rviz2-70]
1684145749.0466745 [ERROR] [rviz2-70]: process has died [pid 7911, exit code -15, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/tofadmin/autoware/install/autoware_launch/share/autoware_launch/rviz/autoware.rviz -s /home/tofadmin/autoware/install/autoware_launch/share/autoware_launch/rviz/image/autoware.png --ros-args -r __node:=rviz2 -p use_sim_time:=True -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7'].
```

nwpuhkp, 2024-05-25:
> I have fixed it by reimporting the assets and enable the read /write features of the fbx model我通过重新导入资产并启用 fbx 模型的读/写功能来修复它

Hello, I encountered a similar problem when running awsim_labs (the latest experimental feature version). Could you please elaborate on your solution?
https://github.com/tier4/AWSIM/issues/76#issuecomment-1408762060
https://github.com/tier4/AWSIM/issues/76#issuecomment-1547749106