autowarefoundation/AWSIM #136: lanelet2 map is unable to load into unity

zhuyingxiang95, 2023-05-30:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other [issues](https://github.com/tier4/AWSIM/issues?q=is%3Aissue) and [discussions Q&A category](https://github.com/orgs/autowarefoundation/discussions/categories/q-a). I found no duplicate issues.
- [X] I'm convinced that this is not my fault but a bug.

### Description

hello guys, I have a lanelet2 map which is converted from another map format. I have tried to import and run  it in Autoware. it works fine. when i tried to load it into unity. it failed and the error is  : an item with the same key is already added.  I guess its something wrong between lanelet2 container script and the lanelet2 file itself. I will appreciate if somesone could  give me some help.

### Expected behavior

lanelet2 map successfully loaded

### Actual behavior

cant load lanelet2 map into unity editor

### Steps to reproduce

load lanelet2 map in a customized scene

zhuyingxiang95, 2023-05-30:
[image]
here is the error screenshot.

zhuyingxiang95, 2023-05-30:
@shmpwk could you please have a look into this issue ?

zhuyingxiang95, 2023-05-30:
this issue is fixed by switching all the node ID ,Lanelets ID into positive. and i guess the code of lanelets_map.cs has a bug in dealing with the negative ID of nodes ,ID of lanelets.  
[image]
I think here we need a setter function to set the value of count , if we don't do that, this.count==0 will be always true. and the unique ID will always be 1. This is the reason for the error why it showed the same key has added yet.

zhuyingxiang95, 2023-05-30:
now another issue is the lanelet2 map could not be fully loaded and it shows the following error. The same lanelet 2 map works fine in autoware [image]
here i attached the lanelet2 map i used
[lanelet2_map.zip](https://github.com/tier4/AWSIM/files/11602341/lanelet2_map.zip)

zhuyingxiang95, 2023-05-31:
@mackierx111 could you please have a look into this issue ?

mackierx111, 2023-08-02:
@zhuyingxiang95 
Thanks for the report. Improved lanelet loading. https://github.com/tier4/AWSIM/pull/172

closed 2023-08-22 by mackierx111