autowarefoundation/AWSIM #139: No stable data from lidar

Aysenayilmaz, 2023-06-01:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other [issues](https://github.com/tier4/AWSIM/issues?q=is%3Aissue) and [discussions Q&A category](https://github.com/orgs/autowarefoundation/discussions/categories/q-a). I found no duplicate issues.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The lidar on the vehicle does not work in the sim environment I created.

### Expected behavior

Lidar provides stable and uninterrupted data.

### Actual behavior

In Unity editor, I built my own environment with the car "**Lexus RX450h 2015 Sample Sensor**", which is ready in Awsim, and I was able to run the simulation. 

When I run the simulation, the topic information I get with the "_ros2 topic list_" command:
_/awsim/ground_truth/vehicle/pose
/clicked_point
/clock
/control/command/control_cmd
/control/command/emergency_cmd
/control/command/gear_cmd
/control/command/hazard_lights_cmd
/control/command/turn_indicators_cmd
/goal_pose
/initialpose
/middle_left/velodyne_points
/parameter_events
/rosout
/sensing/camera/traffic_light/camera_info
/sensing/camera/traffic_light/image_raw
/sensing/gnss/pose
/sensing/gnss/pose_with_covariance
/sensing/imu/tamagawa/imu_raw
/sensing/lidar/top/pointcloud_raw
/sensing/lidar/top/pointcloud_raw_ex
/tf
/tf_static
/vehicle/status/control_mode
/vehicle/status/gear_status
/vehicle/status/hazard_lights_status
/vehicle/status/steering_status
/vehicle/status/turn_indicators_status
/vehicle/status/velocity_status_

I checked the sensors with the "_ros2 topic echo <topic_name>_" command to make sure that the data is healthy, but I noticed that the lidar data is not coming. I did not encounter this issue in the **AWSIM_v1.1.0** demo simulation. I also checked the scripts of lidar, but I couldn't see any difference with nishishinjuku scene. 
[image]

[image]

When I checked the data coming from sensors such as camera, IMU, I could not see a problem with them. But the data flow in the topics "_/sensing/lidar/top/pointcloud_raw"_ and "_/sensing/lidar/top/pointcloud_raw_ex_" from Lidar does not flow when the simulation starts. When you move the vehicle a little bit, there is data flow, but then it goes again.

### The commands working in the terminals in the videos I have uploaded are as follows:
**top left window ->  _ros2 topic list_
top right window -> _ros2 topic echo /sensing/camera/traffic_light/image_raw_
lower left window -> _ros2 topic echo /sensing/lidar/top/pointcloud_raw_
lower right window -> _ros2 topic echo /sensing/imu/tamagawa/imu_raw_**

https://github.com/tier4/AWSIM/assets/108546951/5ee64dbb-b39c-4a8b-9eef-01f39b97beaf

**( Youtube link in case the video can't be played -> https://youtu.be/LrLF8JbqSrI )**

In the video, no data is coming from the lidar. When you move the car a little further, the data starts coming, but after a while it goes again. In the demo simulation, lidar data was coming even when the car was at the starting point without moving at all.
_( My computer is struggling a bit while running the simulation, so I'm sorry I couldn't drive properly. )_

### Steps to reproduce

1. Create a new simulation environment.
2. Build the simulation.
3. Run the simulation and echo the lidar data. 

### Versions

**_Versions_**
- Ubuntu 22.04.2 LTS
- ROS2, Humble
- Unity editor, 2021.1.7f1

**_Pc specs:_**
- Intel® Core™ i7-6700K CPU @ 4.00GHz × 8
- NVIDIA Corporation GA104 [GeForce RTX 3070]

Aysenayilmaz, 2023-06-01:
- As I suspected that the reason for Lidar not functioning might be the environment, I designed an environment from scratch with only a plane and a cube. I built the simulation, and the data flow and Lidar rings were successfully generated.

[image]

- Upon noticing that Lidar was able to detect the cubes, I proceeded to add large cubes in front, back, and on the sides of the vehicle in the previous environment. Despite these modifications, I encountered the issue of not receiving any data.

[image]

- In order to prevent the vehicle from falling, I placed a cube underneath it, intentionally adjusting its position so that it would not interfere with the Lidar rings. No other elements were added to the environment. I did not receive any Lidar data or rings.

[image]

**In every scenario where there was data flow in the Lidar data, the Lidar rings were also visible. However, in some instances where the Lidar seemed to be functioning in my environment, even though there was Lidar data coming through, the Lidar rings were not appearing.**

prybicki, 2023-06-05:
Hello @Aysenayilmaz. Thank you for the report.

This intermittent data flow has never happened before; I am very curious about the reason. My hypothesis is that there may be a mismatch between the configuration of RGL Scene Manager deciding what components are used in raytracing (colliders or meshes) vs. what is provided on the scene.

I could debug it for you and share my insights. Would it be possible to share the scene with me? If you'd prefer, you can email it directly to me at `piotr.rybicki@robotec.ai`

Aysenayilmaz, 2023-06-15:
- Easyroad (the asset i used for roads) and terrain seem to be the reason why Lidar doesn't work in the my environment. When I remove the roads and run the sim when the terrain and other objects remain, I can get the lidar rings and their data. Lidar points do not fall on the terrain because the lidar does not support the terrain object. 

[image]

- I made sure the lidar ring didn't fall on the terrain. There are rings on the walls, but not on the terrain.

[image]

- When I remove the Terrain and the roads and other objects remain, I can't get the lidar rings and data. The reason for this is that even though there is a mesh filter in easyroad, it cannot detect it. To make sure of this, I made the mesh filter the same as the 'plane' object in all roads and, run the sim and the rings came. 

[image]

- I found something about setting up mesh filter for Terrain, I'll try it, if not, maybe it can in Blender. There is a discussion of someone having the same issue with Terrain as I am, I'll be following that as well.  **(https://github.com/tier4/AWSIM/issues/145)**

_- I'm looking into what I can do to fix the roads. There is a mesh filter, but there is something wrong with its path or its files._

Aysenayilmaz, 2023-06-19:
Finally, Lidar point cloud falls on both terrain and roads. 

[image]

The problem about roads seems to be about the hierarchical in unity, not the absence of mesh. I found it with a bit of trial and error, but having the roads under a parent causes meshes to not be read.

In the asset I used, there were all the paths in the environment under a parent, although all of them had their own mesh, when I prefab and run the simulation in this way, I was getting an error that there was no mesh.

It shouldn't be like in the picture. 

[image]

Aysenayilmaz, 2023-07-10:
**I forgot to mention how it should be in my previous comment. The hierarchy should be as shown in the attached image.**

[image]

- I used the _EasyRoad_ asset to create the roads. While creating the roads, this asset creates two child objects under the RoadNetwork parent: connection points and regular roads. I solved the problem I encountered by removing the roads from under the parent object.This problem might be resolved in the future, but since I have found a solution, I am closing the issue I opened.

closed 2023-07-10 by Aysenayilmaz