autowarefoundation/AWSIM #145: Lidar not hitting Unity Terrain

Ushtarador, 2023-06-13:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other issues and no duplicate issues were found.

### Description

Hey all,
I've created some custom offroad terrain using the Unity Terrain editor, but unfortunately the lidar does not register hits on the terrain. My terrain has a Terrain Collider, but I guess that doesn't seem to work - can anyone suggest a smart workaround for this problem?

Ushtarador, 2023-06-15:
Just a small update, I currently worked around the issue by exporting my terrain as a mesh to a .obj file, and then reimporting it, overlaying it with the terrain and assigning it an invisible material. It works for now but I really don't like it ^^

mackierx111, 2023-06-22:
Currently, [RGL](https://github.com/RobotecAI/ros2-for-unity) cannot take terrain into account. It has not yet been determined whether RGL will support it in the future.

msz-rai, 2023-07-31:
Good news! We are currently working to support terrain by LiDARs (by `RGLUnityPlugin`). I am sharing some early results. The feature will be merged into the main branch soon.

[image]
[image]

jediofgever, 2023-08-09:
> Just a small update, I currently worked around the issue by exporting my terrain as a mesh to a .obj file, and then reimporting it, overlaying it with the terrain and assigning it an invisible material. It works for now but I really don't like it ^^

That is hardly a remedy, I did the same temporarily, but what about other objects, e.g trees or plants ? Great news that they are working toward it.

Jakub-Krakowiak, 2023-09-08:
Good news, terrain is now supported as [our pr](https://github.com/tier4/AWSIM/pull/173) has been merged.

closed 2023-09-11 by msz-rai