autowarefoundation/AWSIM #155: Error when trying to read Point Cloud Data

Ushtarador, 2023-06-21:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other issues and no duplicate issues were found.

### Description

Hey all,

I got another problem, and it would be great to get some input on this. I'd like to publish the point cloud data in a ROS1 message, so unfortunately I can't take advantage of the existing ROS2 implementation (unless I'm missing something?). 

 This is the code I wrote:
```
public class RGLLidarROS_PointCloudPublisher : MonoBehaviour
    {        
        private LidarSensor lidarSensor;
        private RGLNodeSequence rglSubgraphUnity2Ros1;
        // Start is called before the first frame update
        void Start()
        {
            lidarSensor = GetComponent<LidarSensor>();

            rglSubgraphUnity2Ros1 = new RGLNodeSequence().AddNodePointsTransform("UNITY_TO_ROS", Matrix4x4.identity);
            lidarSensor.ConnectToLidarFrame(rglSubgraphUnity2Ros1);

            lidarSensor.onNewData += publishPointCloudMessage;                                    
        }

        private void publishPointCloudMessage()
        {
            Vector3[] pointCloud = new Vector3[0];
            rglSubgraphUnity2Ros1.GetResultData<Vector3>(ref pointCloud);
            //write pointCloud into a ROS1 message
        }
    }

```

Unfortunately, I'm getting this error when calling GetResultData:
[image]

If anyone could point me in the right direction on how to directly extract the point cloud data that would be great!

msz-rai, 2023-06-26:
Hey @Ushtarador 
Thank you for the message.

Your code is almost correct. One thing that has been missed is setting the output node to the `RGLNodeSequence`. It it required to inform RGL what fields are expected to produce them. An single field can be requested (with `NodePointsYield`) or multiple fields that will be converted into binary format (with `NodePointsFormat`).

In your case, if you want to get XYZ data, you need to add `AddNodePointsYield` as follows:
`rglSubgraphUnity2Ros1 = new RGLNodeSequence().AddNodePointsTransform("UNITY_TO_ROS", Matrix4x4.identity);
rglSubgraphUnity2Ros1.AddNodePointsYield("GET_XYZ", RGLField.XYZ_F32);`

Please let me know, if it helped you.

Ushtarador, 2023-06-28:
Hey @msz-rai,
Ah I've missed that in the other code, adding the yield node worked, thanks a lot!

Cheers, Christoph

closed 2023-07-31 by msz-rai