autowarefoundation/AWSIM #165: Support for Terrain in Unity Simulation Project

Aysenayilmaz, 2023-07-19:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other issues and no duplicate issues were found.

### Description

Currently, when I create an environment in a regular Unity project and include a terrain object with grass, trees, and ground textures using the Terrain Tool, I am unable to use this terrain in a simulation project that runs on HDRP due to compatibility issues. There is another person using Terrain and having problems ( https://github.com/tier4/AWSIM/issues/145  ).
_(Photos are my environment that I created with the terrain tool.)_

[image]

[image]

Supporting Unity Terrain  in simulation projects would be highly valuable for developers who want to create realistic and visually appealing environments. The ability to utilize HDRP's advanced rendering capabilities in conjunction with Unity Terrain features like grass, trees, ground texture and real-life pointcloud shape would significantly enhance the overall quality and immersion of simulation experiences.

I kindly request that the development team considers adding this feature to allow seamless integration of Unity Terrain in simulation projects, providing developers with a more comprehensive toolkit for creating high-quality simulations.

mackierx111, 2023-07-20:
AWSIM does not yet have full terrain support, including LiDAR Simulation.
Also, is it correct that the terrain compatibility discussed in this issue is the difference between Unity's standard project and the HDRP project?
We recommend creating terrains in HDRP projects in advance. If you really want to import terrains created in the past, you may be able to use them by converting the standard shader material to HDRP/Lit after fbx conversion.

Aysenayilmaz, 2023-07-20:
Since the standard Unity project uses URP and the HDRP project, there are some differences between them. But for example both projects allow the creation of trees and terrain textures etc. 

I checked the shader of my terrain, and it appears to be using HDRP/Lit shader.

[image]

Jakub-Krakowiak, 2023-09-08:
Good news, terrain is now supported as [our pr](https://github.com/tier4/AWSIM/pull/173) has been merged.

closed 2023-09-11 by Aysenayilmaz