autowarefoundation/AWSIM #182: How to run AWSIM and Autoware on two computers?

lawbreaker2022, 2023-08-25:
### Checklist

- [X] I've read the [contribution guidelines](https://tier4.github.io/AWSIM/DeveloperGuide/HowToContribute/).
- [X] I've searched other issues and no duplicate issues were found.

### Description

Nice work! This simulator works with Autoware better than Carla.
I felt that one computer is overloaded when running AWSIM and Autoware together.
How to set the network if i want to run AWSIM and Autoware on two computers?
Thanks.

yuasabe, 2023-08-28:
Hi @lawbreaker2022 !

I've looked at running Autoware and AWSIM on two separate PCs and have documented the process here: 
https://yuasabe.com/2023-08-28-awsim-dual-pc/

Please try this method and feel free to drop more questions here if necessary!

Autostone-c, 2023-08-31:
> Hi @lawbreaker2022 !
> 
> I've looked at running Autoware and AWSIM on two separate PCs and have documented the process here: https://yuasabe.com/2023-08-28-awsim-dual-pc/
> 
> Please try this method and feel free to drop more questions here if necessary!

hi , does it can work on two machines(autoware on jetson Orin(ubungut20.04 python3.8) and awsim on x86 workstation(ubuntu22.04 python3.10))? have you try?

lawbreaker2022, 2023-08-31:
> hi , does it can work on two machines(autoware on jetson Orin(ubungut20.04 python3.8) and awsim on x86 workstation(ubuntu22.04 python3.10))? have you try?

I haven't tried it yet. I'm a bit busy these two weeks, will check with it later.

Autostone-c, 2023-09-02:
> Hi @lawbreaker2022 !
> 
> I've looked at running Autoware and AWSIM on two separate PCs and have documented the process here: https://yuasabe.com/2023-08-28-awsim-dual-pc/
> 
> Please try this method and feel free to drop more questions here if necessary!

hi，i try this work on two machines(autoware on jetson Orin(ubungut20.04 python3.8) and awsim on x86 workstation(ubuntu22.04 python3.10)), but when i run autoware, i see the follow error
[image]
I am checking the code. Do you have any good suggestions？

Autostone-c, 2023-09-02:
> I haven't tried it yet. I'm a bit busy these two weeks, will check with it later.

i am also try this work. maybe we can share our idea

yuasabe, 2023-09-03:
I think it can't find the network interface `en1` with IP address of `10.10.12.90`. Can you run `ip a` to show what network interfaces there are in your computer?

The network interface and IP address you specify here must be the one for the connection between the Jetson Orin and x86 workstation.

closed 2023-10-12 by mackierx111