autowarefoundation/AWSIM #187: Unable to record LiDAR topics suddenly

Yujisuper, 2023-09-04:
Hello,
One day, while I was attempting to retrieve data from AWSIM, I suddenly encountered an issue where I couldn't record any LiDAR topics, even though I hadn't experienced any problems before. I can no longer see the LiDAR lines on the screen either. On the console, I encountered the following 4 error messages.

1.
```
You should not source ROS2 in 'ros2-for-unity' standalone build. Plugin might not work correctly.
UnityEngine.Debug:LogError (object)
ROS2.ROS2ForUnity:CheckIntegrity () (at Assets/Ros2ForUnity/Scripts/ROS2ForUnity.cs:188)
ROS2.ROS2ForUnity:.ctor () (at Assets/Ros2ForUnity/Scripts/ROS2ForUnity.cs:294)
ROS2.ROS2UnityCore:.ctor () (at Assets/Ros2ForUnity/Scripts/ROS2UnityCore.cs:55)
AWSIM.SimulatorROS2Node:Initialize () (at Assets/AWSIM/Scripts/ROS/SimulatorROS2Node.cs:29)
```

2.
```
RobotecGPULidar library cannot be found!
UnityEngine.Debug:LogError (object)
RGLUnityPlugin.RGLNativeAPI:.cctor () (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNativeAPI.cs:163)
RGLUnityPlugin.RGLNodeSequence:AddNodeRaysFromMat3x4f (string,UnityEngine.Matrix4x4[]) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNodeSequence.cs:79)
RGLUnityPlugin.LidarSensor:Awake () (at Assets/RGLUnityPlugin/Scripts/LidarSensor.cs:96)
```

3.
```
DllNotFoundException: RobotecGPULidar
RGLUnityPlugin.RGLNativeAPI.NodeRaysFromMat3x4f (System.IntPtr& node, UnityEngine.Matrix4x4[] rays) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNativeAPI.cs:284)
RGLUnityPlugin.RGLNodeSequence.AddNodeRaysFromMat3x4f (System.String identifier, UnityEngine.Matrix4x4[] rays) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNodeSequence.cs:79)
RGLUnityPlugin.LidarSensor.Awake () (at Assets/RGLUnityPlugin/Scripts/LidarSensor.cs:96)

```
4.
```
DllNotFoundException: RobotecGPULidar
RGLUnityPlugin.RGLNativeAPI.NodePointsTransform (System.IntPtr& node, UnityEngine.Matrix4x4 transform) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNativeAPI.cs:319)
RGLUnityPlugin.RGLNodeSequence.AddNodePointsTransform (System.String identifier, UnityEngine.Matrix4x4 transform) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNodeSequence.cs:123)
AWSIM.RglLidarPublisher.Awake () (at Assets/AWSIM/Scripts/Sensors/LiDAR/RglLidarPublisher.cs:52)
```

My PC environment is as follows:

CPU: Core i9-13900K
GPU: GeForce RTX 4090
GPU Driver: 535.104.05

I have made several attempts to reinstall Ubuntu 22, GPU drivers, ROS2 Humble, and AWSIM, but I still couldn't record. I would greatly appreciate it if you could provide some insight into what might be causing this problem.

[image]
[image]
[image]
[image]

mackierx111, 2023-09-11:
To publish ros2 from UnityEditor, you need to start UnityEditor from the shell where ros2 is sourced. please check it

Yujisuper, 2023-09-13:
Thank you for your reply.
I have confirmed that "source /opt/ros/humble/setup.bash" is written in the .bashrc file and that ros2 is working in the test on the following URL: [https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html#try-some-examples](url)

ismetatabay, 2023-11-15:
@Yujisuper Hi, I solved the same problem by removing the ROS 2 sourcing in the .bashrc file. I think RGLUnityPlugin doesn't work when we source the ROS 2 already. It seems #203 PR is beneficial for users to understand the reason behind the similar problems.

cc: @msz-rai

msz-rai, 2023-11-30:
@mackierx111 @Yujisuper 
I believe the issue can be closed.

Yujisuper, 2024-02-22:
@mackierx111 @msz-rai @ismetatabay 
Sorry for the late response.
I have confirmed that AWSIM works fine on v1.2.0.
Thank you for your help in resolving this issue.

closed 2024-02-22 by Yujisuper