autowarefoundation/AWSIM #199: AutowareSimulation Scene Keeps Crashing

rzninvo, 2023-09-17:
I have downloaded the most recent AWSIM source code and my Unity version is 2021.1.7f1. When I run AutowareSimulation.unity I get this error and my unity crashes:

[image]

rzninvo, 2023-09-17:
## Quick-Fix 
Found a workaround by changing the Mesh Source of the Scene Manager to the **Only Colliders** option.
[image]

msz-rai, 2023-09-18:
This crash is the result of two things:
1. Some game objects had lost their meshes.
2. A bug in `RGLUnityPlugin` that processes invalid objects (with null meshes) endlessly.

For the second point, I have prepared [a PR](https://github.com/tier4/AWSIM/pull/200) that hopefully will be merged soon.

closed 2023-09-21 by rzninvo