autowarefoundation/AWSIM #201: setup tutorial does not work

Autostone-c, 2023-09-20:
i am run run AWSIM(x86 workstation) and Autoware(Jetson Orin) on two computers.

it can work since i run AWSIM_demo.x86_64. but can not communication with autoware when i run AWSIM with Unity Hub.
My steps are as follows：

1：Install the Unity Hub on Linux
[image]

2：Set up the AWSIM project according to the official tutorial
https://tier4.github.io/AWSIM/GettingStarted/SetupUnityProject/

Environment variables and configuration files have not been changed。

anybody has good idea? I guess the tutorial is missing some steps

whbcumtb, 2023-10-07:
Have you played your AWSIM project?

Autostone-c, 2023-10-09:
> Have you played your AWSIM project?

ye，i have already played.I don't think I have added ROS2 related components, but according to the tutorial, I don't seem to need any additional operations

closed 2025-08-07 by mackierx111