autowarefoundation/AWSIM #204: Question about setup a new Pose

yifanfish233, 2023-09-28:
Hi,
I'm runningthe simulation on Ubuntu 20.04 with AWSIM v1.0.1, ROS2 galastic.

I noticed that based on the totural, everything goes fine on my computer, however, once I chose to pick a different position by clicking the "2D Pose Estimate" button, the car position in the AWSIM remain the same.(both the camera image I received from topic and lidar pointcloud). but in rviz, the car moves to correct postion on the map.

I can run without issue by launch the AWSIM and autoware, then setup the goal position, but I can't setup the initial pose correctly. Is there something I didn't setup correctly?

Thanks for the time!

whbcumtb, 2023-11-10:
The correct operation flow is following: 1) start the AWSIM; 2) launch autoware.Then, localization will be initialized automatically in RVIZ. If the car pose is not correctly, you can improve it by clicking the "2D Pose Estimate" button; 3) Setup a goal by clicking the "2D Goal Pose" button. Then the car will move. Good luck!

mackierx111, 2023-11-28:
galactic is not recommended. ros2 humble and AWSIM v1.1.0 is recommended.

closed 2024-03-14 by yifanfish233