autowarefoundation/AWSIM #215: Missing ground truth pose topic

VietPT3502, 2023-10-29:
when i open AWSIM_demo.x86_64 and run ros2 topic list, Only the following topic is return:
/clock
/control/command/control_cmd
/control/command/emergency_cmd
/control/command/gear_cmd
/control/command/hazard_lights_cmd
/control/command/turn_indicators_cmd
/parameter_events
/rosout
/sensing/camera/traffic_light/camera_info
/sensing/camera/traffic_light/image_raw
/sensing/gnss/pose
/sensing/gnss/pose_with_covariance
/sensing/imu/tamagawa/imu_raw
/sensing/lidar/top/pointcloud_raw
/sensing/lidar/top/pointcloud_raw_ex
/vehicle/status/control_mode
/vehicle/status/gear_status
/vehicle/status/hazard_lights_status
/vehicle/status/steering_status
/vehicle/status/turn_indicators_status
/vehicle/status/velocity_status

I dont see ground truth topic in docs which are /awsim/ground_truth/vehicle/pose

Autostone-c, 2023-10-30:
> when i open AWSIM_demo.x86_64 and run ros2 topic list, Only the following topic is return: /clock /control/command/control_cmd /control/command/emergency_cmd /control/command/gear_cmd [...]

hi，i have no /sensing/lidar/top/pointcloud_raw
/sensing/lidar/top/pointcloud_raw_ex topic, do you know why?

VietPT3502, 2023-10-30:
i follow the quick start demo in official docs and get everything well except for the ground_truth topic. maybe you have conflict ros. Im using ros2 humble ubuntu 22.04 and AWSIM_v1.1.0

Autostone-c, 2023-10-30:
> i follow the quick start demo in official docs and get everything well except for the ground_truth topic. maybe you have conflict ros. Im using ros2 humble ubuntu 22.04 and AWSIM_v1.1.0

How did you solve the aforementioned problem（ground_truth  topic）

VietPT3502, 2023-11-01:
> > i follow the quick start demo in official docs and get everything well except for the ground_truth topic. maybe you have conflict ros. Im using ros2 humble ubuntu 22.04 and AWSIM_v1.1.0
> 
> How did you solve the aforementioned problem（ground_truth topic）

i haven't solve it yet.

mackierx111, 2023-11-28:
Sorry, the `/awsim/ground_truth/vehicle/pose` is not published in the demo binary (v1.1.0) that is currently distributed. It will be reflected in the next version.
If necessary, you can build the main branch with UnityEditor and it will be published.

VietPT3502, 2024-01-08:
Hello @mackierx111, sorry for bothering you but i want to know how to calculate the lateral and longitudinal error based on predicted pose and ground truth pose. Here is my example code but i dont think i correctly calculate lateral and longitudinal error.
import threading
import time
import math
import rclpy
from rclpy.node import Node
from geometry_msgs.msg import PoseStamped
from nav_msgs.msg import Odometry

class LocalizationSubscriber(Node):
    def __init__(self):
        super().__init__('localization_subscriber')
        self.subscription_localization = self.create_subscription(
            Odometry,
            '/localization/kinematic_state',
            self.localization_callback,
            10  # QoS profile depth
        )
        self.subscription_ground_truth = self.create_subscription(
            PoseStamped,
            '/awsim/ground_truth/vehicle/pose',
            self.ground_truth_callback,
            10  # QoS profile depth
        )

        # Initialize variables
        self.gt_x, self.gt_y, self.gt_z = 0.0, 0.0, 0.0
        self.x, self.y, self.z = 0.0, 0.0, 0.0
        self.total_error_sum = 0.0
        self.total_error_x,  self.total_error_y,  self.total_error_z = 0.0, 0.0, 0.0
        self.num_errors = 0
        self.max_error_x, self.max_error_y, self.max_error_z = 0.0, 0.0, 0.0
        self.error_x, self.error_y, self.error_z = 0.0, 0.0, 0.0
        self.max_error = 0
        # Flag to indicate if error calculation should continue
        self.calculate_error = False

    def localization_callback(self, msg):
        position = msg.pose.pose.position
        self.x, self.y, self.z = position.x, position.y, position.z

        if self.calculate_error:
            self.calculate_error_metrics()
            self.update_max_errors()
    def ground_truth_callback(self, msg):
        position = msg.pose.position
        self.gt_x, self.gt_y, self.gt_z = position.x, position.y, position.z

    def calculate_error_metrics(self):
        self.error_x = self.x - self.gt_x
        self.error_y = self.y - self.gt_y
        self.error_z = self.z - self.gt_z
        error_magnitude = abs(math.sqrt(self.error_x**2 + self.error_y**2 + self.error_z**2))
        self.max_error = max(self.max_error, error_magnitude)
        self.total_error_sum += error_magnitude
        self.total_error_x += abs(self.error_x)
        self.total_error_y += abs(self.error_y)
        self.total_error_z += abs(self.error_z)
        self.num_errors += 1
    def update_max_errors(self):
        self.max_error_x = max(self.max_error_x, abs(self.error_x))
        self.max_error_y = max(self.max_error_y, abs(self.error_y))
        self.max_error_z = max(self.max_error_z, abs(self.error_z))
    def calculate_and_print_mean_error(self):
        if self.num_errors > 0:
            mean_error_x = self.total_error_x / self.num_errors
            mean_error_y = self.total_error_y / self.num_errors
            mean_error_z = self.total_error_z / self.num_errors
            mean_total_error = self.total_error_sum / self.num_errors

            print(f"Mean Error in x: {mean_error_x} meters")
            print(f"Mean Error in y: {mean_error_y} meters")
            print(f"Mean Error in z: {mean_error_z} meters")
            print(f"Mean Total Error: {mean_total_error} meters")

            print(f"Max Error in x: {self.max_error_x} meters")
            print(f"Max Error in y: {self.max_error_y} meters")
            print(f"Max Error in z: {self.max_error_z} meters")
            print(f"Max Total Error: {self.max_error} meters")
        else:
            print("No errors recorded.")

    def start_error_calculation(self):
        self.calculate_error = True
        print("Error calculation started. Press 'q' to stop and calculate mean error.")

    def stop_error_calculation(self):
        self.calculate_error = False
        self.calculate_and_print_mean_error()
        self.reset()
        print("Error calculation stopped.")
    def reset(self):
        self.gt_x, self.gt_y, self.gt_z = 0.0, 0.0, 0.0
        self.x, self.y, self.z = 0.0, 0.0, 0.0
        self.total_error_sum = 0.0
        self.total_error_x,  self.total_error_y,  self.total_error_z = 0.0, 0.0, 0.0
        self.num_errors = 0
        self.max_error_x, self.max_error_y, self.max_error_z = 0.0, 0.0, 0.0
        self.error_x, self.error_y, self.error_z = 0.0, 0.0, 0.0
        self.max_error = 0
def user_input_thread(node):
    while True:
        user_input = input("Press 'q' to start/stop error calculation: ")
        if user_input.lower() == 'q':
            if node.calculate_error:
                node.stop_error_calculation()
            else:
                node.start_error_calculation()
        time.sleep(0.1)

def main():
    rclpy.init()
    node = LocalizationSubscriber()

    # Start a separate thread for user input
    input_thread = threading.Thread(target=user_input_thread, args=(node,), daemon=True)
    input_thread.start()

    # Main thread for spinning the node
    rclpy.spin(node)
    rclpy.shutdown()

if __name__ == '__main__':
    main()