autowarefoundation/AWSIM #232: Set main branch to linear history

xmfcx, 2023-12-19:
In Autoware repositories we protect the main branch with the `Require linear history ` rule to keep main branch less complicated.

- https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/managing-protected-branches/about-protected-branches#require-linear-history

Steps to enable it:
- https://github.com/tier4/AWSIM/settings/branches
- main -> Edit -> Check `Require linear history` -> Save changes

Some reasoning:
https://stackoverflow.com/a/76321945/4520129

> - **No code changes are hidden inside merge commits as conflict resolution.** Each diff belongs to single commit with its message explaining the reason behind the change.
> - **Conflicts are conceptually easier to resolve** during rebase than during merge, as the resolution takes place in a "context" of single commit (as opposed to entire branch). This reduces potential conflict resolution mistakes.

I understand if you don't want it too, it is a bit subjective.

mackierx111, 2023-12-21:
Thanks for the suggestion! good idea. I have applied the changes.

closed 2023-12-21 by mackierx111

xmfcx, 2023-12-21:
Thank you so much!