autowarefoundation/AWSIM #251: AWSIM Crashes with Autoware.Universe

Sam827-r, 2024-01-22:
Hi all,
Im trying to run the AWSIM demo with Autoware.universe .
AWSIM  gets Segmentation fault (core dumped)  each time when autoware gets launch.

Ref link: https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/#localhost-settings

Machine specs are as follows:

- Ubuntu 22.04.3 LTS
- Linux 6.5.0-14--generic  x86_64
- Processor:AMD® Ryzen 5 3600 6-core processor × 12
- GPU: NVIDIA Corporation TU116 [GeForce GTX 1660 SUPER]
-  NVIDIA-SMI 545.23.08 
- ROS 2 Humble
- awsim-stable branch

mackierx111, 2024-02-26:
AWSIM's LiDAR simualtion utilizes ray tracing and requires an RTX graphics card

doganulus, 2024-02-27:
Are we sure about that RTX requirement? Software fallback would handle ray tracing even without dedicated RT hardware. Performance may be degraded but would not result in a segmentation fault. I think a driver mismatch is more likely, as suggested here: https://github.com/tier4/AWSIM/issues/264. 

But downgrading the driver is not a solution. That could break the user system easily. A better investigation is needed for this issue.

closed 2024-03-12 by mackierx111