autowarefoundation/AWSIM #260: Update AWSIM to the next Unity LTS version

xmfcx, 2024-02-05:
Latest [Unity LTS(Long Term Support) version](https://unity.com/releases/editor/qa/lts-releases) to date is: `2022.3.19f1`

Current AWSIM Unity Version: `2021.1.7f1`

Some of the assets we use such as [CityGen3D](https://assetstore.unity.com/packages/tools/terrain/citygen3d-162468) requires the latest version of the Unity Editor.

And this requirement will only grow larger as the time goes by.

I would like to discuss the issues and roadblocks that are preventing us from moving to the next LTS for the AWSIM project.

Aysenayilmaz, 2024-02-05:
I pulled the current AWsim repository to do a simple test and opened it in Unity version 2022.3.19f1. I didn't see any noticeable issues for now. I tested autonomous driving on the default map as well.

**Pc specs:**

NVIDIA GeForce RTX 4080
AMD® Ryzen 9 5900x 12-core processor × 24

 https://youtu.be/NmCHRW5yCWM

mackierx111, 2025-08-21:
In v2.0.0, the unity version has been upgraded to Unity6.

closed 2025-08-21 by mackierx111