autowarefoundation/AWSIM #264: Unable to start on ubuntu 20.04

is-whale, 2024-02-08:
Unable to start ,due to error displayed on the terminal during Ubuntu 20.04 operation
 whale@pop-os  ~/simluation/AWSIM_v1.2.0  ./AWSIM_v1.2.0.x86_64
Set current directory to /home/whale/simluation/AWSIM_v1.2.0
Found path: /home/whale/simluation/AWSIM_v1.2.0/AWSIM_v1.2.0.x86_64
[1]    18415 segmentation fault (core dumped)  ./AWSIM_v1.2.0.x86_64

21ejy, 2024-02-17:
This issue might be caused by an excessively high version of the NVIDIA driver. There have been instances where a too high Nvidia driver version leads to a Segmentation Fault. In such cases, it's recommended to downgrade the Nvidia driver version to a lower one (version 525 is recommended). You can check your current NVIDIA driver version by running the command `nvidia-smi` in your terminal.

is-whale, 2024-02-19:
OK, I will try to do it. Think you!

doganulus, 2024-02-27:
Why does it fail with the latest released versions of Nvidia drivers? Those drivers are already tested when it comes to our distros.

frankwang98, 2024-06-20:
ubuntu22.04 also appears, nvidia-smi:535