autowarefoundation/AWSIM #265: Intensity values in PointCloud2 topics are zero.

Osal22, 2024-02-12:
Ubuntu= 22.04
AWSIM= 1.2.0
ROS= Humble

topic names I tested published by AWSIM:

/sensing/lidar/top/pointcloud_raw
/sensing/lidar/top/pointcloud_raw_ex

I can not see intensity values other than zero.

Osal22, 2024-02-14:
Thank you! got it.

closed 2024-02-14 by Osal22

ceccocats, 2024-02-20:
> @Osal22
> 
> The feature of simulating intensity in AWSIM is supported, however it requires the materials in the 3D model to have the information about intensity. In the Nishinjuku environment used in release 1.2.0 it's not included for now.
> 
> While intensity is not critical for Autoware's algorithms, it's not the highest priority now to add this information into the Nishinjuku 3D model.

Hi, can you share how to configure a 3d model to have intensity information?
Thanks