autowarefoundation/AWSIM #276: Some questions about AWSIM and ROS

artsung410, 2024-03-12:
1. Looking at the execution method of Quit Start Demo, it seems to get the source code of git and execute it, is it correct?

2. Is there another version for developer debugging other than Quit Start Demo?

3. Can I modify the received source code and use it on my local PC, and if so, to what level?

4. If I look at the source code part "/src" when running Quit Start Demo, it seems that there are only codes corresponding to messaging, settings, and API, but are the other codes downloaded in the form of libraries? Or do I get them as source code, and if so, where is the source code located?

5. Is there any logic or server that receives sensing or command related data and eco-tests that part of the code itself? (This may be unnecessary, but is there any part of the code that can determine if it worked correctly for the data sent?)

6. Is there a separate part for checking the data sent or received, or checking the information processed, in the form of sending and receiving data without using unity to render?

7. Is it possible to change the type or number of sensors that can be added to a single car in AWSim? (The documentation seems to indicate that it is, but is it?)

8. In the case of AWSIm, are the messages exchanged for sensors based on ROS-based topics? What type of messages are exchanged?

Kalpit-Vadnerkar, 2024-03-12:
I would too like to know the answers to these questions.
Also, Is there a way to change the rostopic names that AWSIM publishes to, even the autoware ones?
my goal it to add some noise to the sensor data being sent to the autoware stack.

mackierx111, 2024-03-12:
@Kalpit-Vadnerkar 
> Is there a way to change the rostopic names that AWSIM publishes to, even the autoware ones?

yes. For example, if you want to change the topic to publish on the IMU sensor, just change the Topic in ImuRos2Publisher.
[image]

Kalpit-Vadnerkar, 2024-03-13:
Is the screenshot from the Unity project? I will have to make changes to the unity project and rebuild it. Am I thinking about this the correct way?

mackierx111, 2024-03-25:
@Kalpit-Vadnerkar 
sorry for late reply. 
Yes. Currently we need to do various configurations in UnityEditor. This is because we value extensibility over convenience. But in the future we may make sensors and ros2 configurable with Json and other tools.

closed 2025-08-21 by mackierx111