autowarefoundation/AWSIM #324: Weather Simulation for AWSIM + Autoware.Universe

TangChenTom, 2024-06-15:
Dear Developer,
     Is there any plan to deploy the weather simulation in AWSIM? Different weather would affect the sensors in Autoware, which is important for real scenarios. Some simulator like Carla has weather editing function. It would be awesome if AWSIM has the similar function.
     
Sincerely thanks for developing AWSIM!

mackierx111, 2024-06-19:
Weather simulation is one of the hot topics of our development. However, we believe that it is meaningless if the weather changes only apparently. We need to simulate the impact that weather has on each sensor and vehicle.
We have not yet added a major feature called weather, but we are adding the ability to simulate noise for each component.

For example, a LiDAR sensor can apply Gaussian noise to a point cloud. (https://tier4.github.io/AWSIM/Components/Sensors/LiDARSensor/LiDARSensor/#lidar-sensor-script)
Vehicles can be made slippery by applying road surface slip to the wheels.
(https://tier4.github.io/AWSIM/Components/Vehicle/CustomizeSlip/)

TangChenTom, 2024-06-20:
Thanks for your reply! Does it mean that AWSIM supports adding noise to sensors or adjusting the parameters of the vehicle's component for the simulation of weather impact?

mackierx111, 2024-06-25:
Yes, it is. But not all. For example, the camera sensor does not yet support snow weather. Also, I think each user will be satisfied with different accuracy.

The features that will be added in AWSIM will be implemented for use in the development of Autoware.