autowarefoundation/AWSIM #330: Request for improvement of CPU load increase due to deadlock resolution

HayatoF-68, 2024-07-03:
I would like to ask for improvement of the CPU load when applying pull requests [“feat(RandomTraffic): improve yielding rules to avoid deadlocks #243”](https://github.com/tier4/AWSIM/pull/243).

I performed Random simulations immediately after installation of “v.1.2.1 with deadlock removal applied” and “v.1.2.0 before the application was applied”.
Max Vehicle Count was set to 200.

I performed the experiment on the following devices:
・OS: Ubuntu 22.04.3 LTS
・Processor: 13th Gen Intel® Core™ i9-13900K × 32
・Memory: 128GB
・Hardware: Dell Inc. Precision 3660
・Graphic: NVIDIA Corporation GA102GL [RTX A6000] / NVIDIA RTX A6000/PCIe/SSE2A6000

Results:
In v.1.2.1, a significant increase in processing volume was observed after a certain amount of time had passed. The fps also dropped from around 100 at first to single digits.
It was found that the majority of the causes were due to processing related to the newly added "Cognition.CheckRightOfWay"(which is not dominant in v.1.2.0).

Could you please fix it so that the CPU load is reduced even when the Max Vehicle Count is high?

[image]
Profiler v.1.2.1

[image]
Task details v.1.2.1

[image]
Profiler v.1.2.0

[image]
Task details v.1.2.0

lchojnack, 2024-09-05:
Hi @HayatoF-68 

I believe that the issue has been resolved in the PR https://github.com/tier4/AWSIM/pull/337

HayatoF-68, 2024-09-23:
Thank you for resolving this issue.
It worked fine!

closed 2024-09-23 by HayatoF-68