autowarefoundation/AWSIM #336: Frequent crashes of Unity Editor 2021.1.7f1 When trying to open AWSIM project

ajaycrytek, 2024-07-25:
Steps taken to reproduce the issue
1. Start Unity Hub on Ubuntu 22.04 (Inside Autoware Dokcer Container as well as Outside on the Ubuntu Desktop)
2. Sign in and start the Project AWSIM
3. Unity Hub loads and Crashes with the following Pop Up
[image]

This Behaviour is too frequent

Other Points to be Noted:
1. Followed the instructions on https://tier4.github.io/AWSIM/GettingStarted/SetupUnityProject/
2. installed libssl, vulkan drivers. using Nvidia drivers 555.52.04, cuda_12.3.r12.3, RTX 4070 8GB Mobile GPU, Ubuntu 22.04, 32GB RAM

The Unity Editor opened and It worked for the first time like a charm and after system restart It started to crash continously and randomnly it start to work fine

lchojnack, 2024-09-05:
Hi @ajaycrytek 

I had a similar issue. Downgrading the nvidia-driver to 550 resolved the issue in my case.

lchojnack, 2024-09-06:
@ajaycrytek 

Another fix that helped in my case was to clean the project with the command `git clean -xdf` and re-import of unitypackage

closed 2025-08-21 by mackierx111