autowarefoundation/AWSIM #338: the higest speed is too small

cq1528, 2024-07-25:
I encountered the following problem in the simulation of AWSIM software v1.1 using autoware.unniverse. I modified the velocimety_limit parameter in the file to 40m/s for universe/autoware.universe/planning/motion-velocity_smother/config/default_motion-velocity_smother. param.yaml. However, in the actual simulation, the vehicle only reached a maximum speed of 16.88m/s (60km/h). Can you tell me how to solve this problem and which parameter still needs to be modified?

I performed the experiment on the following devices:
・OS: Ubuntu 22.04.3 LTS
・Processor: 13th Gen Intel® Core™ i9-13900K × 32
・Memory: 128GB
・Graphic: NVIDIA A5000

[image]

ahmedius2, 2025-01-14:
Have you tried modifying the following file?
src/launcher/autoware_launch/autoware_launch/config/planning/scenario_planning/common/common.param.yaml

mackierx111, 2025-08-21:
The default value of Autoware's maximum speed is 15 km/h. If you want more speed, you can modify it by entering the following command in the autoware directory.
```
sed -i 's/max_vel: 4.17/max_vel: 22.2/' \
./src/launcher/autoware_launch/autoware_launch/config/planning/scenario_planning/common/common.paramyaml
```

see also https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/#4-run-awsim-and-autoware