autowarefoundation/AWSIM #347: Segmentation fault (core dumped)

hamitkoculu, 2024-08-22:
```
hamitkoculu@hamitkoculu:~/AWSIM_v1.2.3$ ./AWSIM.x86_64 
Set current directory to /home/hamitkoculu/AWSIM_v1.2.3
Found path: /home/hamitkoculu/AWSIM_v1.2.3/AWSIM.x86_64
Segmentation fault (core dumped)
```

I'm using a laptop. I have Nvidia RTX 2060 (mobile) graphics card. I installed nvidia-driver-535 because my graphics card does not support the 525 version recommended in the documentation. (I checked here https://www.nvidia.com/en-us/drivers/). For this reason, when I run Autoware, I get a Segmentation fault (core dumped) error and AWSIM shuts down. It is recommended to install nvidia-driver-525 as a solution, but as I said, I cannot do this. Does this mean I can't work on the computer? I would be happy if you help.

lchojnack, 2024-09-05:
Hi @hamitkoculu 

Please check how much GPU memory your Nvidia RTX 2060 (mobile) has (`nvidia-smi` command).
From my experience, the `Segmentation fault (core dumped)` is likely caused by insufficient GPU memory available.

hamitkoculu, 2024-09-05:
First of all thank you for your reply @lchojnack,

This is the output of `nvidia-smi` command:

`+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 560.28.03              Driver Version: 560.28.03      CUDA Version: 12.6     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 2060        Off |   00000000:01:00.0  On |                  N/A |
| N/A   53C    P8             13W /   90W |     265MiB /   6144MiB |      2%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      1144      G   /usr/lib/xorg/Xorg                            106MiB |
|    0   N/A  N/A      1458      G   /usr/bin/gnome-shell                          140MiB |
+-----------------------------------------------------------------------------------------+`

I am asking just to be sure, It's not possible for me to work with this graphics card, right?

lchojnack, 2024-09-05:
From what I see you have around 6 GB of GPU memory, so it should be possible to run AWSIM and Autoware.

This is my `nvidia-smi` when running AWSIM and Autoware, so it takes around 6 GB so you are close to the limits.
```
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 560.35.03              Driver Version: 560.35.03      CUDA Version: 12.6     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 3070 ...    Off |   00000000:01:00.0 Off |                  N/A |
| N/A   62C    P0             56W /  115W |    5925MiB /   8192MiB |     36%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      1581      G   /usr/lib/xorg/Xorg                             15MiB |
|    0   N/A  N/A      5392    C+G   .../Hub/Editor/2021.1.7f1/Editor/Unity       4821MiB |
|    0   N/A  N/A      6615      C   ...p_components/component_container_mt        620MiB |
|    0   N/A  N/A      7097      C   ...p_components/component_container_mt        196MiB |
|    0   N/A  N/A      7159      C   ...p_components/component_container_mt        196MiB |
+-----------------------------------------------------------------------------------------+

```

What you can try to do:
- If your laptop has internal GPU, then you can select `prime-select on-demand` which means only intensive tasks will be used by NVIDIA GPU, as AWSIM and Autoware. In that case, gnome-shell will use internal gpu.
- From the Unity Editor you can select `very low` quality. Edit->Project Settings->Quality then select `Very Low`. Also from the same settings page, you can reduce `Texture Quality` to `Eighth Res`

hamitkoculu, 2024-09-05:
> From what I see you have around 6 GB of GPU memory, so it should be possible to run AWSIM and Autoware.
>
> This is my `nvidia-smi` when running AWSIM and Autoware, so it takes around 6 GB so you are [...]

My laptop has an internal GPU, but it is already disabled in the BIOS. This means I'm only using NVIDIA GPU. As you said, it is very likely that I am getting this problem with a sudden increase in memory usage since it is running at the full limit. Upgrading my graphics card seems like the only solution. Thank you for your information about your `nvidia-smi`. Best wishes, @lchojnack.

closed 2024-09-05 by hamitkoculu